ngscopeclient/scopehal-apps #586: Docs out of date ? (or I'm missing something)

artgodwin, 2023-04-14:
The manual states :

The corresponding minimum hardware requirement is an AMD Radeon HD 5000, NVIDIA
GeForce GTX 6xx series discrete GPU, or Intel Skylake or newer integrated GPU, plus suitably
up-to-date drivers. On Linux with recent Mesa, Intel integrated GPUs as old as Ivy Bridge may be
used.

However, on a Radeon 5450 HD, the open source drivers report Opengl 3.3 and the AMD drivers refuse to install because, on Debian 11, the Xserver is 1.20 and the AMD drivers only support 1.10. Multiple reports exist that these AMD drivers  (catalyst 15.9) will not report on other systems running 1.19 and this appears to have been a problem for some time.

AMD did update some drivers and released amdgpu-pro-18.20-673703-ubuntu-18.04 but this appears to only cover later GPUS.

I'm pretty naive with video cards : I have no gaming requirement and so I try to buy cheap fanless cards but unless I've missed installing something (a later Mesa ?) I think the 5450 is now too old to be useful. Updated  suggestion of minimal version would be welcome.

azonenberg, 2023-06-13:
Ngscopeclient now requires Vulkan 1.0 (preferably 1.1) and does not use OpenGL. There's updated documentation in the works with new minimum requirements..

artgodwin, 2023-07-08:
Builds and executes on Debian 11 and vulkan 1.3.224 with above-mentioned video card.
Built as described in manual (required fresh clone - a pull on top of previous clone broke in  MetricsDialog)
testing with generate and Agilent DSO7014A
hangs and/or segfaults in operation - I will join IRC to investigate

azonenberg, 2023-07-30:
Closing as ngscopeclient now has its own manual. Open a separate ticket against scopehal-docs if you find any additional issues.

closed 2023-07-30 by azonenberg