ngscopeclient/scopehal-apps #589: ngscopeclient Windows crash when loading with "File => RecentFiles" (work fine when using "File => Open Offline...")

bvernoux, 2023-05-27:
Tested with ngscopeclient 0.1-e261779 (Commit 25 May 2023) Windows11 Pro Ryzen 9 7950X3D/RTX4070
https://github.com/glscopeclient/scopehal-apps/commit/e2617799fdf28b9984db4b87c7b2ac3224e2e3c5

Data used see [CH569_SerDes_180Mbps_data_scopesession.zip](https://github.com/glscopeclient/scopehal-apps/files/11581252/CH569_SerDes_180Mbps_data_scopesession.zip)

### No Crash Recent Files(with debug trace):
```
ngscopeclient.exe --debug --stdout-only --trace MainWindow
File => Recent Files => D:\glscopeclient\
[CH569_SerDes_180Mbps_data_scopesession.zip](https://github.com/glscopeclient/scopehal-apps/files/11581247/CH569_SerDes_180Mbps_data_scopesession.zip)
CH569_SerDes_180Mbps.scopesession
```
- No issue all work fine

### Crash Recent Files(without debug trace)
```
ngscopeclient.exe --debug --stdout-only
File => Recent Files => D:\glscopeclient\CH569_SerDes_180Mbps.scopesession
Warning: Attempted to bind an empty buffer
```
- Crash of ngscopeclient

### Never Crash Open Offline
```
ngscopeclient.exe
ngscopeclient.exe --debug
ngscopeclient.exe --debug --stdout-only
File => Open Offline... => D:\glscopeclient\CH569_SerDes_180Mbps.scopesession
```
- No issue all work fine

bvernoux, 2023-06-09:
For information I cannot reproduce that issue with latest ngscopeclient https://github.com/glscopeclient/scopehal-apps/commit/335945e1990297b80d6bec0806f267d123d47638

azonenberg, 2023-06-13:
Closing for now then, reopen if it comes back.

closed 2023-06-13 by azonenberg