ngscopeclient/scopehal-apps #596: ngscopeclient preferences.yml font path shall use relative path by default

bvernoux, 2023-06-10:
Today ngscopeclient preferences.yml use full path for all font for everything (label_font ...) even when created by default using fonts from ./fonts/xxx
So it will be better to use relative path by default at least as when you launch ngscopeclient from an other path the old one which is a full path is not found and it prevent to ngscopeclient start with an error
How to fix it: 
- Just use relative path "./font/"
- In case font path is wrong not found it shall rollback to default font with relative font path

azonenberg, 2023-06-11:
Relative paths doesn't sound like a great idea, since we need to allow for things like running a dev build in your working copy followed by installing system wide, and not have them stepping on each other accessing stuff in /usr/share/fonts or similar.

The better solution is for the font loading code to detect failure to load a font file and fall back to a default font if the supplied filename is nonexistent or malformed. This will recover without crashing in the rare case of a font you have selected becoming unavailable, while also ensuring determinism in other cases.

bvernoux, 2023-06-11:
Please check the https://github.com/glscopeclient/scopehal/pull/779 that clearly solve any issue with relative or full path (if relative path is not available it do like before scanning all dir for full path ...)

bvernoux, 2023-06-12:
Fixed in commit [f236404](https://github.com/glscopeclient/scopehal/commit/f236404a40d1219d9475bc8e0390e38c0202004a)

closed 2023-06-12 by bvernoux