ngscopeclient/scopehal-apps #597: ngscopeclient: Black screen and unresponsive when starting on Ubuntu 22.04

vincentbaeten, 2023-06-28:
I can run glscopeclient but ngscopeclient gives me a black screen and is unresponsive. I've tried it with Vulkan 1.3.250 and 1.3.239 which give me both the same output as given below using `export VK_LOADER_DEBUG=error,warn,info,layer,driver`.

I've  not copied the vulkan sdk files into my /usr like the manual says but I've set all the environment variables which I think is enough for testing? This is a bit out of my depth so I do not really know where to go from here.

```
INFO:             Vulkan Loader Version 1.3.250
INFO:             Vulkan Loader Version 1.3.250
LAYER:            Searching for layer manifest files
LAYER:               In following locations:
LAYER:                  /home/vincent/.config/vulkan/implicit_layer.d
LAYER:                  /etc/xdg/xdg-ubuntu/vulkan/implicit_layer.d
LAYER:                  /etc/xdg/vulkan/implicit_layer.d
LAYER:                  /etc/vulkan/implicit_layer.d
LAYER:                  /home/vincent/.local/share/vulkan/implicit_layer.d
LAYER:                  /usr/share/ubuntu/vulkan/implicit_layer.d
LAYER:                  /home/vincent/.local/share/flatpak/exports/share/vulkan/implicit_layer.d
LAYER:                  /var/lib/flatpak/exports/share/vulkan/implicit_layer.d
LAYER:                  /usr/local/share/vulkan/implicit_layer.d
LAYER:                  /usr/share/vulkan/implicit_layer.d
LAYER:                  /var/lib/snapd/desktop/vulkan/implicit_layer.d
LAYER:               Found the following files:
LAYER:                  /usr/share/vulkan/implicit_layer.d/VkLayer_MESA_device_select.json
INFO:             Found manifest file /usr/share/vulkan/implicit_layer.d/VkLayer_MESA_device_select.json (file version "1.0.0")
DRIVER:           Searching for driver manifest files
DRIVER:              In following locations:
DRIVER:                 /home/vincent/.config/vulkan/icd.d
DRIVER:                 /etc/xdg/xdg-ubuntu/vulkan/icd.d
DRIVER:                 /etc/xdg/vulkan/icd.d
DRIVER:                 /etc/vulkan/icd.d
DRIVER:                 /home/vincent/.local/share/vulkan/icd.d
DRIVER:                 /usr/share/ubuntu/vulkan/icd.d
DRIVER:                 /home/vincent/.local/share/flatpak/exports/share/vulkan/icd.d
DRIVER:                 /var/lib/flatpak/exports/share/vulkan/icd.d
DRIVER:                 /usr/local/share/vulkan/icd.d
DRIVER:                 /usr/share/vulkan/icd.d
DRIVER:                 /var/lib/snapd/desktop/vulkan/icd.d
DRIVER:              Found the following files:
DRIVER:                 /usr/share/vulkan/icd.d/radeon_icd.x86_64.json
DRIVER:                 /usr/share/vulkan/icd.d/intel_icd.x86_64.json
DRIVER:                 /usr/share/vulkan/icd.d/lvp_icd.x86_64.json
DRIVER:           Found ICD manifest file /usr/share/vulkan/icd.d/radeon_icd.x86_64.json, version "1.0.0"
DEBUG | DRIVER:   Searching for ICD drivers named /usr/lib/x86_64-linux-gnu/libvulkan_radeon.so
DRIVER:           Found ICD manifest file /usr/share/vulkan/icd.d/intel_icd.x86_64.json, version "1.0.0"
DEBUG | DRIVER:   Searching for ICD drivers named /usr/lib/x86_64-linux-gnu/libvulkan_intel.so
DRIVER:           Found ICD manifest file /usr/share/vulkan/icd.d/lvp_icd.x86_64.json, version "1.0.0"
DEBUG | DRIVER:   Searching for ICD drivers named /usr/lib/x86_64-linux-gnu/libvulkan_lvp.so
DRIVER:           Searching for driver manifest files
DRIVER:              In following locations:
DRIVER:                 /home/vincent/.config/vulkan/icd.d
DRIVER:                 /etc/xdg/xdg-ubuntu/vulkan/icd.d
DRIVER:                 /etc/xdg/vulkan/icd.d
DRIVER:                 /etc/vulkan/icd.d
DRIVER:                 /home/vincent/.local/share/vulkan/icd.d
DRIVER:                 /usr/share/ubuntu/vulkan/icd.d
DRIVER:                 /home/vincent/.local/share/flatpak/exports/share/vulkan/icd.d
DRIVER:                 /var/lib/flatpak/exports/share/vulkan/icd.d
DRIVER:                 /usr/local/share/vulkan/icd.d
DRIVER:                 /usr/share/vulkan/icd.d
DRIVER:                 /var/lib/snapd/desktop/vulkan/icd.d
DRIVER:              Found the following files:
DRIVER:                 /usr/share/vulkan/icd.d/radeon_icd.x86_64.json
DRIVER:                 /usr/share/vulkan/icd.d/intel_icd.x86_64.json
DRIVER:                 /usr/share/vulkan/icd.d/lvp_icd.x86_64.json
DRIVER:           Found ICD manifest file /usr/share/vulkan/icd.d/radeon_icd.x86_64.json, version "1.0.0"
DEBUG | DRIVER:   Searching for ICD drivers named /usr/lib/x86_64-linux-gnu/libvulkan_radeon.so
DRIVER:           Found ICD manifest file /usr/share/vulkan/icd.d/intel_icd.x86_64.json, version "1.0.0"
DEBUG | DRIVER:   Searching for ICD drivers named /usr/lib/x86_64-linux-gnu/libvulkan_intel.so
DRIVER:           Found ICD manifest file /usr/share/vulkan/icd.d/lvp_icd.x86_64.json, version "1.0.0"
DEBUG | DRIVER:   Searching for ICD drivers named /usr/lib/x86_64-linux-gnu/libvulkan_lvp.so
LAYER:            Searching for layer manifest files
[... 288 lines cut ...]
INFO | DRIVER:               [1] AMD Unknown (RADV REMBRANDT)
INFO | DRIVER:         Sorted order:
INFO | DRIVER:               [0] AMD Unknown (RADV REMBRANDT)  
INFO | DRIVER:               [1] llvmpipe (LLVM 15.0.7, 256 bits)  
DEBUG | DRIVER:   Copying old device 0 into new device 0
DEBUG | DRIVER:   Copying old device 1 into new device 1
INFO | DRIVER:    linux_read_sorted_physical_devices:
INFO | DRIVER:         Original order:
INFO | DRIVER:               [0] llvmpipe (LLVM 15.0.7, 256 bits)
INFO | DRIVER:               [1] AMD Unknown (RADV REMBRANDT)
INFO | DRIVER:         Sorted order:
INFO | DRIVER:               [0] AMD Unknown (RADV REMBRANDT)  
INFO | DRIVER:               [1] llvmpipe (LLVM 15.0.7, 256 bits)  
DEBUG | DRIVER:   Copying old device 0 into new device 0
DEBUG | DRIVER:   Copying old device 1 into new device 1
INFO | LAYER:     Failed to find vkGetDeviceProcAddr in layer "libVkLayer_MESA_device_select.so"
DRIVER | LAYER:   vkCreateDevice layer callstack setup to:
DRIVER | LAYER:      <Application>
DRIVER | LAYER:        ||
DRIVER | LAYER:      <Loader>
DRIVER | LAYER:        ||
DRIVER | LAYER:      <Device>
DRIVER | LAYER:          Using "AMD Unknown (RADV REMBRANDT)" with driver: "/usr/lib/x86_64-linux-gnu/libvulkan_radeon.so"
Killed
```

vincentbaeten, 2023-06-28:
I am using a AMD Ryzen 7 PRO 6850U with Radeon Graphics and vulkaninfo gives me the following output https://gist.github.com/vincentbaeten/61ae0b4662e6d3ec0a13d18ded0d0bb9

bvernoux, 2023-06-28:
Unfortunately today in actual ngscopeclient version AMD Ryzen Integrated GPU are not correctly supported for unknown reason.
- Note: I have also tried on my Ryzen 9 7950X3D integrated GPU with similar issue as in theory it shall work with Vulkan anyway in my case as I have a GeForce RX4070 which is used by default it was required to change the source code/rebuild to force to recognize only the AMD Ryzen Integrated GPU and so maybe I have probably different issue as you in that test...

What is recommended until it is fixed is to use an external GFX card which have the minimal specifications to run correctly so not something which is >10years old...

hansemro, 2023-08-12:
Notice that the g_mainWindow.render queue is selecting the wrong queue family without the graphics bits. This is likely the cause of rendering issues. I presume something is wrong with QueueManager::GetQueueWithFlags.

hansemro, 2023-08-12:
Indeed, if I replace the following (at https://github.com/glscopeclient/scopehal/blob/master/scopehal/QueueManager.h#L139):
```
{ return GetQueueWithFlags(vk::QueueFlagBits::eGraphics | vk::QueueFlagBits::eTransfer, name); }
```

with:
```
{ return GetQueueWithFlags(vk::QueueFlagBits::eGraphics, name); }
```

I can run ngscopeclient on a laptop on AMD integrated GPU (3750H + Vega 8) without the GPU driver crashing.

azonenberg, 2023-08-12:
Interesting. The graphics queue should also have the transfer flag set (so you can move data around that you're drawing).

I'm pretty sure the QueueManager code was written by @lainy so maybe she can weigh in on if this is a bug there or somewhere else?

hansemro, 2023-08-12:
Looking at https://github.com/glscopeclient/scopehal/blob/96a5e6360a39d1b040e3fd5aea513d98b9b2f4b2/scopehal/QueueManager.cpp#L182-L184:
```
		//Skip if flags don't match
		if(!(m_queues[i].Flags & flags))
			continue;
```

we can see that partially matching flags will not be skipped. If we were to check if the flags match exactly, then we can correctly select the queue.

```
		//Skip if flags don't match
		if((m_queues[i].Flags & flags) != flags)
			continue;
```

azonenberg, 2023-08-12:
Yeah that looks like a bug. @vincentbaeten @bvernoux can you check if the above change fixes the problem on your systems?

vincentbaeten, 2023-08-12:
@azonenberg This does seem to solve it using the latest commit https://github.com/glscopeclient/scopehal-apps/commit/37aa92be2a73f916b1446caca705190fb898df2d. I haven't tested it extensively but I can add my scope and trigger it.

azonenberg, 2023-08-12:
That's good enough for me. @hansemro plase send a PR and let's get this bug squished.

hansemro, 2023-08-12:
PR submitted: https://github.com/glscopeclient/scopehal/pull/790

closed 2023-08-12 by azonenberg