ngscopeclient/scopehal-apps #612: Rigol: Floating point exception

penguin42, 2023-09-20:
Connecting to a DS1102Z-E Rigol via LAN, clicked on the single shot button and got:
Warning: Ran out of data after 0 points 
Warning: Attempted to bind an empty buffer
Floating point exception (core dumped)

#0  0x00000000008cc374 in WaveformArea::RasterizeAnalogOrDigitalWaveform(std::shared_ptr<DisplayedChannel>, vk::raii::CommandBuffer&, bool)
    (this=0x33729f0, channel=std::shared_ptr<DisplayedChannel> (use count 3, weak count 0) = {...}, cmdbuf=..., clearPersistence=false)
    at /discs/more/git/scopehal-apps/src/ngscopeclient/WaveformArea.cpp:1603
#0  0x00000000008cc374 in WaveformArea::RasterizeAnalogOrDigitalWaveform (this=0x33729f0, channel=..., cmdbuf=..., clearPersistence=false)
    at /discs/more/git/scopehal-apps/src/ngscopeclient/WaveformArea.cpp:1603
1603    int64_t offset_samples = (offset - data->m_triggerPhase) / data->m_timescale;

(gdb) p *data
$2 = {_vptr.WaveformBase = 0x9af840 <vtable for UniformWaveform<float>+16>, m_timescale = 0, m_startTimestamp = 16951
70282,
  m_startFemtoseconds = 23995876312255, m_triggerPhase = 0, m_flags = 0 '\000', m_revision = 0}

#1  0x00000000008cd42f in WaveformArea::RenderWaveformTextures(vk::raii::CommandBuffer&, std::vector<std::shared_ptr<DisplayedChannel>, std::allocator<std::shared_ptr<DisplayedChannel> > >&, bool)
    (this=0x33729f0, cmdbuf=..., chans=std::vector of length 1, capacity 1 = {...}, clearPersistence=clearPersistence@entry=false)
    at /discs/more/git/scopehal-apps/src/ngscopeclient/WaveformArea.cpp:1545
#2  0x00000000008db504 in WaveformGroup::RenderWaveformTextures(vk::raii::CommandBuffer&, std::vector<std::shared_ptr<DisplayedChannel>, std::allocator<std::shared_ptr<DisplayedChannel> > >&, bool)
    (this=<optimized out>, cmdbuf=..., channels=std::vector of length 1, capacity 1 = {...}, clearPersistence=clearPersistence@entry=false)
    at /discs/more/git/scopehal-apps/src/ngscopeclient/WaveformGroup.cpp:164
#3  0x0000000000801436 in MainWindow::RenderWaveformTextures(vk::raii::CommandBuffer&, std::vector<std::shared_ptr<DisplayedChannel>, std::allocator<std::shared_ptr<DisplayedChannel> > >&) (this=<optimized out>, cmdbuf=<optimized out>, channels=<optimized out>)
    at /discs/more/git/scopehal-apps/src/ngscopeclient/MainWindow.cpp:434
#4  0x00000000008831cc in Session::RenderWaveformTextures(vk::raii::CommandBuffer&, std::vector<std::shared_ptr<DisplayedChannel>, std::allocator<std::shared_ptr<DisplayedChannel> > >&) (this=<optimized out>, cmdbuf=<optimized out>, channels=<optimized out>)
    at /discs/more/git/scopehal-apps/src/ngscopeclient/Session.cpp:2166
#5  0x00000000008e7edc in RenderAllWaveforms(vk::raii::CommandBuffer&, Session*, std::shared_ptr<QueueHandle>)
    (cmdbuf=..., session=0x3714770, queue=std::shared_ptr<QueueHandle> (use count 5, weak count 0) = {...})
    at /discs/more/git/scopehal-apps/src/ngscopeclient/WaveformThread.cpp:167
#6  0x00000000008e8a34 in WaveformThread(Session*, std::atomic<bool>*) (session=0x3714770, shuttingDown=0x37148a0)
    at /discs/more/git/scopehal-apps/src/ngscopeclient/WaveformThread.cpp:144
#7  0x00007fb1950e31f3 in std::execute_native_thread_routine(void*) (__p=0x3d778e0) at ../../../../../libstdc++-v3/src/c++11/thread.cc:104
#8  0x00007fb194dcb897 in start_thread (arg=<optimized out>) at pthread_create.c:444
#9  0x00007fb194e5261c in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78

azonenberg, 2023-09-20:
Looks like what's happening here is that the scope is doing something the driver doesn't like during acquisition (header_blocksize = 0 in RigolOscilloscope.cpp:848).

The driver fails in a rather unusual way (creates a waveform with zero points, rather than the expected way of sending "no data" which is to set the waveform pointer to null) which then causes a bunch of the GUI logic to choke.

Can you re-run with `--debug --trace SCPISocketTransport` (assuming you're using the LAN transport) on the command line and upload the output? Probably best to make it an attachment as the dump will likely be large.

penguin42, 2023-09-20:
Hmm, please find attached a log generated by:

./src/ngscopeclient/ngscopeclient --debug --trace SCPISocketTransport > log 2>&1

although it ends in:
[SCPISocketTransport::ReadRawData]     Got 250001 bytes
[SCPISocketTransport::SendCommand]     [ronnie] Sending WAV:STAR 22250001
[SCPISocketTransport::SendCommand]     [ronnie] Sending WAV:STOP 22500000
[SCPISocketTransport::SendCommand]     [ronnie] Sending WAV:DATA?
[SCPISocketTransport::ReadRawData]     Got 2 bytes
[SCPISocketTransport::ReadRaw

which makes me wonder whether it's complete.

I restarted ngscopeclient and did it again, but it failed much faster, and this time I left out the redirects, so here is where it ended on my terminal:

Connecting to SCPI device at ronnie:5555
[SCPISocketTransport::SendCommand] [ronnie] Sending *IDN?
[SCPISocketTransport::ReadReply] [ronnie] Got RIGOL TECHNOLOGIES,DS1102Z-E,DS1ZE231706403,00.06.02
[SCPISocketTransport::SendCommand] [ronnie] Sending :WAV:FORM BYTE
[SCPISocketTransport::SendCommand] [ronnie] Sending :WAV:MODE RAW
[SCPISocketTransport::SendCommand] [ronnie] Sending :TIM:VERN ON
[SCPISocketTransport::SendCommand]     [ronnie] Sending :CHAN1:DISP?
[SCPISocketTransport::ReadReply]     [ronnie] Got 1
[SCPISocketTransport::SendCommand]     [ronnie] Sending :CHAN2:DISP?
[SCPISocketTransport::ReadReply]     [ronnie] Got 0
[SCPISocketTransport::SendCommand]     [ronnie] Sending :CHAN1:DISP ON
[SCPISocketTransport::SendCommand] [ronnie] Sending :TRIG:MODE?
[SCPISocketTransport::ReadReply] [ronnie] Got EDGE
[SCPISocketTransport::SendCommand] [ronnie] Sending :TRIG:EDGE:SOUR?
[SCPISocketTransport::ReadReply] [ronnie] Got CHAN1
[SCPISocketTransport::SendCommand] [ronnie] Sending :TRIG:EDGE:LEV?
[SCPISocketTransport::ReadReply] [ronnie] Got 0.000000e+00
[SCPISocketTransport::SendCommand] [ronnie] Sending :TRIG:EDGE:SLOPE?
[SCPISocketTransport::ReadReply] [ronnie] Got POS
[SCPISocketTransport::SendCommand] [ronnie] Sending :TIM:MAIN:OFFS?
[SCPISocketTransport::ReadReply] [ronnie] Got 0.0000000e+00
[SCPISocketTransport::SendCommand] [ronnie] Sending :CHAN1:OFFS?
[SCPISocketTransport::ReadReply] [ronnie] Got 0.000000e+00
[SCPISocketTransport::SendCommand] [ronnie] Sending :CHAN1:RANGE?
[SCPISocketTransport::ReadReply] [ronnie] Got 4.000000e-02
[SCPISocketTransport::SendCommand]         [ronnie] Sending :SING
[SCPISocketTransport::SendCommand]         [ronnie] Sending *WAI
[SCPISocketTransport::SendCommand] [ronnie] Sending :TRIG:STAT?
[SCPISocketTransport::ReadReply] [ronnie] Got STOP
[SCPISocketTransport::SendCommand]     [ronnie] Sending WAV:SOUR CHAN1
[SCPISocketTransport::SendCommand]     [ronnie] Sending WAV:PRE?
[SCPISocketTransport::ReadReply]     [ronnie] Got 0,2,1200,1,5.000000e-11,-3.000000e-08,0,2.000000e-04,0,127
[SCPISocketTransport::SendCommand]     [ronnie] Sending WAV:STAR 1
[SCPISocketTransport::SendCommand]     [ronnie] Sending WAV:STOP 1200
[SCPISocketTransport::SendCommand]     [ronnie] Sending WAV:DATA?
[SCPISocketTransport::ReadRawData]     Got 2 bytes
[SCPISocketTransport::ReadRawData]     Got 9 bytes
    Warning: Ran out of data after 0 points
Warning: Attempted to bind an empty buffer
Segmentation fault (core dumped)

which makes me wonder if it left the scope in an odd state.

penguin42, 2023-09-20:
[log.txt](https://github.com/glscopeclient/scopehal-apps/files/12672769/log.txt)

penguin42, 2023-09-20:
I get the feeling there's something timing going on here;
I broke out tcpdump and got a slightly different failure:
```
[SCPISocketTransport::SendCommand] [ronnie] Sending :CHAN1:RANGE?
[SCPISocketTransport::ReadReply] [ronnie] Got 4.000000e-02
[SCPISocketTransport::SendCommand]         [ronnie] Sending :SING
[SCPISocketTransport::SendCommand]         [ronnie] Sending *WAI
[SCPISocketTransport::SendCommand] [ronnie] Sending :TRIG:STAT?
[SCPISocketTransport::ReadReply] [ronnie] Got STOP
[SCPISocketTransport::SendCommand]     [ronnie] Sending WAV:SOUR CHAN1
[SCPISocketTransport::SendCommand]     [ronnie] Sending WAV:PRE?
[SCPISocketTransport::ReadReply]     [ronnie] Got 0,2,0,1,0.000000,0.000000,0,0.000000,0,0
Floating point exception (core dumped)

penguin42, 2023-09-20:
Oh, I think this is as simple that it hasn't actually triggered yet - there's no data;  and depending on the timing it might be in the middle of the capture.

penguin42, 2023-09-20:
Manually triggering on the scope panel and *then* throwing ngscopeclient at it and it's working.

azonenberg, 2023-09-21:
Very interesting, because it's supposed to be polling the trigger status and not trying to download data until it's actually triggered.

As a guess, this is a bug where it can't tell the difference between "trigger stopped because the scope just booted and has no data" and "trigger stopped because we armed the trigger and then it triggered".

azonenberg, 2023-09-23:
Moving to scopehal as this is a driver bug, zero point waveforms aren't well defined so it's not surprising the GUI chokes if given one.