ngscopeclient/scopehal-apps #618: MSO5000 ngscopeclient: Floating point exception (core dumped)

nm17, 2023-10-03:
Steps to reproduce:
  1. Connect your MSO5000 scope to your PC using USB.
  2. Start NGScopeClient
  3. Connect your oscilloscope using USBTMC.
  4. See that the NGScopeClient seems to work fine.
  5. Press the "Arm the trigger in the normal mode" button.
  6. NGScopeClient crashes

Version: Nightly Github Actions build of cc4335b on Linux (can't compile myself because of #611 )

OS: Ubuntu 23.04
Linux: 6.2.0-33-generic
CPU: Intel Core i5-9300H

Log:
```
OMP_WAIT_POLICY not set to PASSIVE
Re-exec'ing with correct environment
Initializing Vulkan
    VK_KHR_get_physical_device_properties2: supported
    VK_KHR_xcb_surface: supported
    VK_KHR_xlib_surface: supported
    VK_EXT_debug_utils: supported
    Loader/API support available for Vulkan 1.3
    Vulkan 1.2 support available, requesting it
    Initializing glfw 3.3.8 X11 GLX EGL OSMesa clock_gettime evdev shared
    GLFW required extensions:
        VK_KHR_surface
        VK_KHR_xcb_surface
    Physical devices:
        Device 0: Intel(R) UHD Graphics 630 (CFL GT2)
            API version:            0x004030ee (0.1.3.238)
            Driver version:         0x05c00004 (0.23.0.4)
            Vendor ID:              8086
            Device ID:              3e9b
            Device type:            Integrated GPU
            int64:                  yes
            int16:                  yes (allowed in SSBOs)
            int8:                   yes (allowed in SSBOs)
            Max image dim 2D:       16384
            Max storage buf range:  4095 MB
            Max mem alloc:          4095 MB
            Max compute shared mem: 64 KB
            Max compute grp count:  65535 x 65535 x 65535
            Max compute invocs:     1024
            Max compute grp size:   1024 x 1024 x 1024
            Memory types:
                Type 0
                    Heap index: 0
                    Device local
                    Host visible
                    Host coherent
                    Host cached
            Memory heaps:
                Heap 0
                    Size: 11 GB
                    Device local
        Device 1: llvmpipe (LLVM 15.0.7, 256 bits)
            API version:            0x004030ee (0.1.3.238)
            Driver version:         0x00000001 (0.0.0.1)
            Vendor ID:              10005
            Device ID:              0000
            Device type:            CPU
            int64:                  yes
            int16:                  yes (allowed in SSBOs)
            int8:                   yes (allowed in SSBOs)
            Max image dim 2D:       16384
            Max storage buf range:  128 MB
            Max mem alloc:          4095 MB
            Max compute shared mem: 32 KB
            Max compute grp count:  65535 x 65535 x 65535
            Max compute invocs:     1024
            Max compute grp size:   1024 x 1024 x 1024
            Memory types:
                Type 0
                    Heap index: 0
                    Device local
                    Host visible
                    Host coherent
                    Host cached
            Memory heaps:
                Heap 0
                    Size: 2 GB
                    Device local
        Selected device 0
            Queue families (1 total)
                Queue type 0
                    Queue count:          1
                    Timestamp valid bits: 36
                    Graphics
                    Compute
                    Transfer
            Driver: vk::DriverId::eIntelOpenSourceMESA
            Enabling 64-bit float support
            Enabling 64-bit integer support
            Enabling 16-bit integer support
            Enabling 16-bit integer support for SSBOs
            Enabling 8-bit integer support
            Enabling 8-bit integer support for SSBOs
            Device has VK_KHR_shader_non_semantic_info, requesting it
            Device has VK_EXT_memory_budget, requesting it
            Device has VK_EXT_shader_atomic_float, requesting it
            Using type 0 for pinned host memory
            Using type 0 for card-local memory
            Sorted queues:
                Family=0 Index=0 Flags=00000007
            QueueManager creating family=0 index=0 name=g_vkTransferQueue
    
    vkFFT version: 1.3.2
Detecting CPU features...
    * AVX2
    * FMA
    
Using ImGui version 1.89.8
Vulkan driver is Mesa.
Disabling vkSetDebugUtilsObjectNameEXT on VkSurfaceKHR objects to work around driver bug.
Unable to open recently used files list (bad file)
Connecting to SCPI oscilloscope over USBTMC through /dev/usbtmc4
SCPITMCTransport::FlushRXBuffer is unimplemented
Floating point exception (core dumped)
```

azonenberg, 2023-10-21:
Hi, can you try with the latest build and, if it still crashes, provide a backtrace?

nm17, 2023-12-04:
@azonenberg Sorry for the delay) Now it's throwing a `vk::OutOfDeviceMemoryError` under the same circumstances as the floating point error. I have no idea how to proceed forward. I'm  using  proprietary nvidia drivers and a GTX1650

```
terminate called after throwing an instance of 'vk::OutOfDeviceMemoryError'
  what():  vkCreateBuffer: ErrorOutOfDeviceMemory
Aborted (core dumped)
```

nm17, 2023-12-04:
Used version: https://github.com/ngscopeclient/scopehal-apps/actions/runs/7089667570

azonenberg, 2025-02-04:
We've made some improvements to low memory handling recently. Can you try again?

azonenberg, 2025-06-24:
Closing as this is believed to be fixed, please reopen or file a new ticket if it's still happening