ngscopeclient/scopehal-apps #621: Build on macOS with Homebrew-provided Vulkan dependencies

rgov, 2023-10-29:
The build instructions for macOS call for some dependencies to be installed from Homebrew formulae and others to come from the LunarG VulkanSDK.

There are also brew formulae for many components of the VulkanSDK. Just grepping for formulae from the KhronosGroup GitHub organization:

```
glslang
molten-vk
opencl-clhpp-headers
opencl-headers
opencl-icd-loader
shaderc
spirv-cross
spirv-headers
spirv-llvm-translator
spirv-tools
vulkan-extensionlayer
vulkan-headers
vulkan-loader
vulkan-tools
vulkan-validationlayers
```

However I am unsure which VulkanSDK components are actually needed to build ngscopeclient. Could you please clarify?

---

<details>
<summary>Homebrew formula so far</summary>

```
class Ngscopeclient < Formula
  desc "Advanced test and measurement instrument remote control and analysis suite"
  homepage "https://www.ngscopeclient.org/"
  license "BSD-3-Clause"
  head "https://github.com/ngscopeclient/scopehal-apps.git"

  depends_on "catch2"
  depends_on "cmake" => :build
  depends_on "glew"
  depends_on "glfw"
  depends_on "gtk+3"
  depends_on "gtkmm3"
  depends_on "libomp"
  depends_on "pkg-config"
  depends_on "yaml-cpp"

  def install
    ENV["VULKAN_SDK"] = "/opt/VulkanSDK/1.3.231.1/macOS"
    ENV.prepend_path "PATH", Pathname.new(ENV["VULKAN_SDK"])/"bin"

    system "cmake", "-S", ".", "-B", "build", *std_cmake_args
    system "cmake", "--build", "build"
    system "cmake", "--install", "build"
  end

  test do
    system "true"
  end
end
```
</details>

azonenberg, 2023-10-29:
+ @lainy

d235j, 2024-01-22:
I've been working on this. A change to Homebrew will be required as they are currently building shaderc without `BULD_SHARED_LIBS` enabled.

FindVulkan is going away upstream as well, so I'm eliminating the use of its functionality from scopehal / scopehal-apps proper; VkFFT still uses it though.

rgov, 2024-01-22:
Nice. Not sure if you saw the formula I had in the expandable part of the first post, but it's a good starting point.

d235j, 2024-01-28:
This should be fixed in #673.

d235j, 2024-02-10:
Fixed in https://github.com/ngscopeclient/scopehal-apps/pull/673.

closed 2024-02-10 by d235j

rgov, 2024-02-10:
I built `HEAD` successfully from Homebrew, after uninstalling the Vulkan SDK. 

```
  depends_on "shaderc" => :build
  depends_on "glslang"
  depends_on "spirv-tools"
  depends_on "vulkan-headers" => :build
  depends_on "vulkan-loader"

  on_macos
    depends_on "molten-vk"
  end
```

Does that look right? I'm not sure about runtime vs build time dependencies. The formula ideally works on Linux too, but I haven't tried it.

d235j, 2024-02-10:
Yeah I think that looks right. I would test by unlinking the build dependencies and seeing if it runs.

We will still have to deal with #623 and #624 before this formula can be recommended to others. Also, see my comment regarding #622.

rgov, 2024-02-10:
Updated formula above which builds successfully, but unfortunately:

```
$ DYLD_PRINT_LIBRARIES=1 ngscopeclient
dyld[69110]: <5E817763-3CAF-3B26-AED0-C6AEF44F64FD> /opt/homebrew/Cellar/ngscopeclient/HEAD-6d43215/bin/ngscopeclient
dyld[69110]: <0020E098-5381-3D35-8497-6D73AEA3EE9C> /opt/homebrew/Cellar/ngscopeclient/HEAD-6d43215/lib/libscopeprotocols.dylib
dyld[69110]: <1BBDB3AB-5EA4-3F74-93A7-81AE744A17F7> /opt/homebrew/Cellar/libpng/1.6.42/lib/libpng16.16.dylib
dyld[69110]: <1B108106-B69A-36D5-8BFC-838FB2E0F220> /opt/homebrew/Cellar/ngscopeclient/HEAD-6d43215/lib/libscopehal.dylib
dyld[69110]: <4751A4FF-7299-3AA4-B376-1E30311EACE5> /opt/homebrew/Cellar/cairomm@1.14/1.14.5/lib/libcairomm-1.0.1.dylib
dyld[69110]: <93D635A0-A3C4-34FE-97D8-89BA8637730E> /opt/homebrew/Cellar/libsigc++@2/2.10.8/lib/libsigc-2.0.0.dylib
dyld[69110]: <9B361D2C-5963-39FA-9A74-49E16FA1E89A> /opt/homebrew/Cellar/cairo/1.18.0/lib/libcairo.2.dylib
dyld[69110]: <9384219F-C495-3D50-8B8B-DF644C1E4B14> /opt/homebrew/Cellar/yaml-cpp/0.8.0/lib/libyaml-cpp.0.8.0.dylib
dyld[69110]: <190965F7-FB16-319D-A75E-B978DE5AB361> /opt/homebrew/Cellar/libomp/17.0.6/lib/libomp.dylib
dyld[69110]: <6DDE3DA2-CE03-3574-B170-6536E59DC3B9> /opt/homebrew/Cellar/vulkan-loader/1.3.277/lib/libvulkan.1.3.277.dylib
dyld[69110]: <04D26FCB-E108-39DA-9EE5-216D93155667> /opt/homebrew/Cellar/glslang/14.0.0/lib/libglslang.14.0.0.dylib
dyld[69110]: <4581293C-E8D3-37E6-8344-822588487F3C> /opt/homebrew/Cellar/glslang/14.0.0/lib/libSPIRV.14.0.0.dylib
dyld[69110]: <AAACA6D4-25FE-3B21-A204-77C0AB05F7AD> /opt/homebrew/Cellar/spirv-tools/1.3.275.0/lib/libSPIRV-Tools-opt.dylib
dyld[69110]: <1A0E6609-6D68-320C-BB35-00413B85FBAE> /opt/homebrew/Cellar/spirv-tools/1.3.275.0/lib/libSPIRV-Tools.dylib
dyld[69110]: <A2E22F90-7507-3CBB-A6BC-890FC43A1312> /opt/homebrew/Cellar/glfw/3.3.9/lib/libglfw.3.3.dylib
...
libc++abi: terminating due to uncaught exception of type std::runtime_error: Failed to load vulkan library!
```

Backtrace:

```
    frame #9: 0x00000001018b68bc libscopehal.dylib`vk::DynamicLoader::DynamicLoader(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&) + 112
    frame #10: 0x00000001018b672c libscopehal.dylib`vk::raii::Context::Context() + 40
    frame #11: 0x00000001018c8af8 libscopehal.dylib`_GLOBAL__sub_I_VulkanInit.cpp + 28
```

rgov, 2024-02-10:
Ok, formula is working again with Homebrew dependencies after I fixed #688 by including the appropriate RPATH.

rgov, 2024-02-12:
@azonenberg At this point we can pretty much submit this to Homebrew, but I can already predict they'd prefer you tagged a release rather than always installing from `HEAD`.

d235j, 2024-02-12:
@rgov A while ago Homebrew moved to binary-first, so we'll have to wait until there is a release before doing so. That said, we could create a tap. However, there are some issues with installation (e.g. hardcoded paths in #624 and plugin search issues in #623) that have to be fixed first.

rgov, 2024-02-12:
My formula already patches #623 and #624 with workarounds for the Homebrew prefix, so those aren't blockers at this moment.

d235j, 2024-02-20:
@rgov would you like to PR your formula to https://github.com/ngscopeclient/homebrew-ngscopeclient, or should I add it?