ngscopeclient/scopehal-apps #622: macOS: ngscopeclient binary is missing RPATH load command

rgov, 2023-10-29:
I built ngscopeclient on macOS 14.0. The binary links in several libraries using `@rpath`:

```
% otool -L $(which ngscopeclient) | grep @rpath  
	@rpath/libscopeprotocols.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libscopeexports.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libscopehal.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libvulkan.1.dylib (compatibility version 1.0.0, current version 1.3.231)
```

However the binary does not contain any `LC_RPATH` load commands, so `@rpath` is undefined.

```
However the resulting binary cannot be run:

```
% ngscopeclient                              
dyld[45566]: Library not loaded: @rpath/libscopeprotocols.dylib
  Referenced from: <87EE124C-9597-35DF-9F0A-5A19B4B39F9F> /opt/homebrew/Cellar/ngscopeclient/HEAD-6056adc/bin/ngscopeclient
  Reason: no LC_RPATH's found
```

azonenberg, 2023-10-29:
+ @lainy

rgov, 2023-10-29:
I added the CMake argument `-DCMAKE_INSTALL_RPATH="/opt/homebrew/opt/ngscopeclient/lib;/opt/VulkanSDK/1.3.231.1/macOS/lib"` to work around the issue.

azonenberg, 2023-10-29:
Yeah I don't think we currently have install support on macos (i.e. it's assumed you are running from the build directory).

Not surprised there's rough edges if you try to do a systemwide install, but the feedback is welcome as proper installability/packaging is one of the things we're trying to fix before the v0.1 release.

d235j, 2024-01-28:
You probably need `-DCMAKE_INSTALL_RPATH=#{rpath}` in your formula, as many other formulas also have this.

azonenberg, 2025-06-24:
@rgov does this issue still occur with the latest version of https://github.com/ngscopeclient/homebrew-ngscopeclient? If not, let's close this

closed 2025-06-24 by rgov