ngscopeclient/scopehal-apps #623: macOS: Tensorflow assertion failures on startup

rgov, 2023-10-29:
On macOS 14.0, I built ngscopeclient 6056adc479ba2a80531e1b535b7b7052c2a8a153 and got it running with the workarounds covered in other issues plus the environment variables from https://github.com/ngscopeclient/scopehal-docs/pull/58.

When starting up, I get several complaints from Tensorflow, then the program aborts.

```
2023-10-29 14:14:13.425505: W tensorflow/core/common_runtime/input_colocation_exemption_registry.cc:33] Input colocation exemption for op: IdentityN already registered
2023-10-29 14:14:13.430134: E tensorflow/tsl/lib/monitoring/collection_registry.cc:81] Cannot register 2 metrics with the same name: /tensorflow/serving/batching/enable_large_batch_splitting
2023-10-29 14:14:13.433836: E tensorflow/tsl/lib/monitoring/collection_registry.cc:81] Cannot register 2 metrics with the same name: /tensorflow/api/op/using_fake_quantization
2023-10-29 14:14:13.442854: F tensorflow/core/framework/function.cc:2027] Check failed: GetOpGradFactory()->insert({op, func}).second Duplicated gradient for Softmax
zsh: abort      ngscopeclient
```

rgov, 2023-10-29:
No, it looks like somehow it is loading ... every single binary in /opt/homebrew. Impressive.

azonenberg, 2023-10-29:
I think what might be happening is something funky happening with the plugin loading code, where it thinks /opt/homebrew is the plugin search directory so it's trying to dlopen() everything there?

Like I said, system wide installation on macos has never been tested.

azonenberg, 2023-10-29:
+ @lainy  as well since I don't have a mac to test on

rgov, 2023-10-29:
Could you point me to the plugin loading code and what the install paths look like on Linux?

azonenberg, 2023-10-29:
It looks like we have a workaround for this exact issue but /opt/homebrew isn't included.

https://github.com/ngscopeclient/scopehal/blob/master/scopehal/scopehal.cpp?ts=4#L299

rgov, 2023-10-29:
https://github.com/ngscopeclient/scopehal-apps/issues/393

lainy, 2024-10-04:
Closing as this appears to have been fixed at some point. Please re-open if this is incorrect.

closed 2024-10-04 by lainy