ngscopeclient/scopehal-apps #625: macOS: Scale GUI for Retina displays

rgov, 2023-10-29:
The GUI is awkwardly scaled on macOS, where most Apple displays are going to be high DPI ("Retina"). Compare the menu text with the titlebar which is the normal system font size. 

[image]

There's some discussion here: https://github.com/ocornut/imgui/issues/3757

rgov, 2023-10-29:
`GDK_SCALE=2` doesn't do it (#255 etc.)

azonenberg, 2023-10-29:
ngscopeclient doesn't use GTK, so any of the pre-2022 issues about scaling won't be relevant.

Font scaling is calculated based on GetContentScale() from imgui (see MainWindow::UpdateFonts()).

Is it possible there's a double scaling going on where the OS is trying to scale fonts and the application also is, and they're stepping on each other?

+ @lainy @602p

azonenberg, 2023-10-29:
Most UI elements are calculated with sizes based on the font size, so as text size scales everything else should scale to match.

So the root cause is likely the font scaling factor being off somehow.

closed 2024-02-08 by azonenberg

rgov, 2024-02-10:
Looks great, thanks @d235j!

[image]