ngscopeclient/scopehal-apps #626: macOS: glfwGetRequiredInstanceExtensions failed

niklaut, 2023-10-30:
I'm following the install instruction on macOS 13.6 on an ARM64 MBA M2.
I installed the Homebrew and Vulkan SDK successfully. Cloned the scopehal-apps and set the exports correctly (also verified the paths).

```
cmake ../ -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_PREFIX_PATH="/opt/homebrew;/opt/homebrew/opt/libomp"
make -j12
```
experiences multiple errors during build:

```
[ 85%] Linking CXX executable Primitives
ld: warning: ignoring duplicate libraries: '-latk-1.0', '-latkmm-1.6', '-lcairo-gobject', '-lcairomm-1.0', '-lgdk-3', '-lgdkmm-3.0', '-lgio-2.0', '-lgiomm-2.4', '-lglibmm-2.4', '-lgtk-3', '-lgtkmm-3.0', '-lpangomm-1.4', '-lsigc-2.0'
    ERROR: glfwGetRequiredInstanceExtensions failed
...
[ 88%] Linking CXX executable Filters
ld: warning: ignoring duplicate libraries: '-latk-1.0', '-latkmm-1.6', '-lcairo-gobject', '-lcairomm-1.0', '-lgdk-3', '-lgdkmm-3.0', '-lgio-2.0', '-lgiomm-2.4', '-lglibmm-2.4', '-lgtk-3', '-lgtkmm-3.0', '-lpangomm-1.4', '-lsigc-2.0'
    ERROR: glfwGetRequiredInstanceExtensions failed

[100%] Linking CXX executable ngscopeclient
ld: warning: ignoring duplicate libraries: '-lcairo', '-lcairomm-1.0', '-lsigc-2.0'
[100%] Built target ngscopeclient
make: *** [all] Error 2
```

But it does produce a `ngscopeclient` binary, which then fails with the same error.

```
$ ./src/ngscopeclient/ngscopeclient
    ERROR: glfwGetRequiredInstanceExtensions failed
```

I tried googling this error, but my complete lack of graphics understanding (I work on microcontrollers) makes this difficult to fix.
For example: I tried `export VK_ICD_FILENAMES=$VULKAN_SDK/etc/vulkan/icd.d/MoltenVK_icd.json` [from here](https://github.com/glfw/glfw/issues/1299), but that did nothing.

Is this because of my own config? Maybe a conflicting homebrew install?
<details>
<summary>Output of `brew list`</summary>

```
$ brew list
==> Formulae
abseil				fastcdr				gtkmm3				libogg				libzip				openjpeg			sdl2
adwaita-icon-theme		fastdds				gts				libomp				lima				openldap			sigrok-cli
ant				fastddsgen			harfbuzz			libpng				little-cms2			openssl@1.1			six
aom				ffmpeg				hicolor-icon-theme		libqalculate			llvm@16				openssl@3			snappy
aribb24				flac				hidapi				librist				lua				opus				speex
arm-gcc-bin@12			flock				highway				librsvg				luajit				p11-kit				sqlite
arm-gcc-bin@9			fontconfig			hiredis				libsamplerate			lz4				pango				srt
arm-gcc-xpack@12		foonathan-memory		hunspell			libsigc++@2			lzo				pangomm@2.46			svt-av1
asio				freetype			icu4c				libsigrok			m4				pcre				tcl-tk
assimp				frei0r				imath				libsigrokdecode			mavsdk				pcre2				telnet
astyle				fribidi				isl				libslirp			mbedtls				picocom				tesseract
atk				gcc				jasper				libsndfile			md4c				pixman				texinfo
atkmm@2.28			gd				jpeg-turbo			libsodium			meson				pkg-config			theora
autoconf			gdbm				jpeg-xl				libsoxr				mosquitto			protobuf			tinyxml2
automake			gdk-pixbuf			jsoncpp				libssh				mpdecimal			px4-sim-jmavsim			tree
boost				genromfs			kconfig-frontends		libssh2				mpfr				py3cairo			uchardet
brotli				gettext				lame				libtasn1			mpg123				pygobject3			unbound
c-ares				giflib				leptonica			libtiff				mpv				python-certifi			vapoursynth
ca-certificates			glew				libarchive			libtool				mujs				python-setuptools		vde
cairo				glfw				libass				libunibreak			mutagen				python@3.11			webp
cairomm@1.14			glib				libavif				libunistring			ncurses				python@3.12			wget
capstone			glibmm@2.66			libb2				libusb				netpbm				python@3.8			x264
catch2				gmp				libbluray			libuv				nettle				qalculate-gtk			x265
ccache				gnu-sed				libcerf				libvidstab			ninja				qemu				xorgproto
cjson				gnu-tar				libelf				libvmaf				nmap				qt				xvid
cloc				gnuplot				libepoxy			libvorbis			node				qt@5				xz
cmake				gnutls				libevent			libvpx				numpy				radare2				yaml-cpp
colima				gobject-introspection		libftdi				libwebsockets			open-ocd			rav1e				yarn
confuse				gradle				libgit2@1.6			libx11				openblas			re2				yt-dlp
curl				graphite2			libidn2				libxau				opencore-amr			readline			z3
dav1d				graphviz			liblinear			libxcb				openexr				ripgrep				zeromq
dbus				grpc				libmng				libxdmcp			openjdk				rtmpdump			zimg
double-conversion		gsettings-desktop-schemas	libmpc				libxext				openjdk@11			rubberband			zstd
dtc				gtk+3				libnghttp2			libxrender			openjdk@17			rust

==> Casks
1password		bettertouchtool		gitx			istat-menus		multipass		segger-jlink		sublime-text		vlc
aldente			discord			google-drive		iterm2			obsidian		slack			utm			xquartz
android-platform-tools	docker			gtkwave			kicad			saleae-logic		spotify			visual-studio-code	zerotier-one
```

</details>

azonenberg, 2023-10-30:
The errors during build are due to a long-standing (but low priority) bug in which some unit tests try to initialize Vulkan during compilation during what's supposed to just be a pass to enumerate what test cases are available.

The failure with ngscopeclient and the tests is the same, MoltenVK isn't initializing right. Try running ngscopeclient with --debug to dump more information that might help us narrow down the problem.

+ @lainy @602p

niklaut, 2023-10-31:
Hm ok, interesting. If I check the "System Global Installation" checkbox in the Vulkan SDK installer and then not `export` any paths, the build succeeds. Or maybe the reboot fixed it? Well… it's running now, thank you!

<details>
<summary>ngscopeclient --debug output</summary>

```
$ ./src/ngscopeclient/ngscopeclient --debug
OMP_WAIT_POLICY not set to PASSIVE
Re-exec'ing with correct environment
Initializing Vulkan
    VK_KHR_get_physical_device_properties2: supported
    VK_EXT_debug_utils: supported
    Loader/API support available for Vulkan 1.3
    Vulkan 1.2 support available, requesting it
    Initializing glfw 3.3.8 Cocoa NSGL EGL OSMesa dynamic
    GLFW required extensions:
        VK_KHR_surface
        VK_EXT_metal_surface
    Physical devices:
        Device 0: Apple M2
            API version:            0x004020e7 (0.1.2.231)
            Driver version:         0x000027d8 (0.0.2.2008)
            Vendor ID:              106b
            Device ID:              d0603ef
            Device type:            Integrated GPU
            int64:                  yes
            int16:                  yes (allowed in SSBOs)
            int8:                   yes (allowed in SSBOs)
            Max image dim 2D:       16384
            Max storage buf range:  4095 MB
            Max mem alloc:          1024 MB
            Max compute shared mem: 32 KB
            Max compute grp count:  1073741824 x 1073741824 x 1073741824
            Max compute invocs:     1024
            Max compute grp size:   1024 x 1024 x 1024
            Memory types:
                Type 0
                    Heap index: 0
                    Device local
                Type 1
                    Heap index: 0
                    Device local
                    Host visible
                    Host coherent
                    Host cached
                Type 2
                    Heap index: 0
                    Device local
                    Host visible
                    Host cached
                Type 3
                    Heap index: 0
                    Device local
                    Lazily allocated
            Memory heaps:
                Heap 0
                    Size: 24 GB
                    Device local
        Selected device 0
            Queue families (4 total)
                Queue type 0
                    Queue count:          1
                    Timestamp valid bits: 64
                    Graphics
                    Compute
                    Transfer
                Queue type 1
                    Queue count:          1
                    Timestamp valid bits: 64
                    Graphics
                    Compute
                    Transfer
                Queue type 2
                    Queue count:          1
                    Timestamp valid bits: 64
                    Graphics
                    Compute
                    Transfer
                Queue type 3
                    Queue count:          1
                    Timestamp valid bits: 64
                    Graphics
                    Compute
                    Transfer
            Driver: vk::DriverId::eMoltenvk
            Enabling 64-bit integer support
            Enabling 16-bit integer support
            Enabling 16-bit integer support for SSBOs
            Enabling 8-bit integer support
            Enabling 8-bit integer support for SSBOs
            Device has VK_KHR_portability_subset, requesting it
            Device has VK_KHR_push_descriptor, requesting it
            Device has VK_EXT_memory_budget, requesting it
            Using type 1 for pinned host memory
            Using type 0 for card-local memory
            Sorted queues:
                Family=0 Index=0 Flags=00000007
                Family=1 Index=0 Flags=00000007
                Family=2 Index=0 Flags=00000007
                Family=3 Index=0 Flags=00000007
            QueueManager creating family=0 index=0 name=g_vkTransferQueue

    vkFFT version: 1.3.2
Detecting CPU features...
QueueManager creating family=1 index=0 name=g_mainWindow.render
Using ImGui version 1.89.8
QueueManager creating family=2 index=0 name=FilterGraphExecutor[0].queue
QueueManager creating family=3 index=0 name=FilterGraphExecutor[5].queue
Unable to open recently used instruments file
Unable to open recently used files list (bad file)
ngscopeclient(51832,0x1e31ee080) malloc: tiny_free_list_remove_ptr: Internal invariant broken (next ptr of prev): ptr=0x13864a510, prev_next=0x0
ngscopeclient(51832,0x1e31ee080) malloc: *** set a breakpoint in malloc_error_break to debug
[1]    51832 abort      ./src/ngscopeclient/ngscopeclient --debug
```
</details>