ngscopeclient/scopehal-apps #634: Unclear how to delete a filter graph node

juh2600, 2023-11-20:
Expected behavior: To delete a node, either click the node and hit Delete or Backspace on the keyboard, or right/double-click the node and choose a "Delete" option.

Observed behavior:
- Delete and Backspace keys don't delete nodes
- No apparent menu option to delete them
- Manual [Ch. 8 Filter Graph Editor](https://www.ngscopeclient.org/manual/FilterGraphEditor.html#x10-490008) doesn't seem to have anything about removing or deleting nodes

azonenberg, 2023-11-20:
As of now, filter objects and channels are reference counted and automatically deleted when the last consumer (measurement display, sink / export filter, waveform view, or filter/channel input) is removed. So we don't allow deletion of a node explicitly, you simply stop using it and it goes away.

I agree this is potentially not the best UX, but it's not a trivial problem: if you delete the last viewport showing a waveform and don't have the graph view open or any other filters consuming its output, it seems intuitive that the filter generating that waveform should disappear and stop consuming resources.

But if you have the graph view open you might want to keep it around so you can use it somewhere else. We also have some ugly hacks in the code around export filters (which have nothing consuming their output, as there are no output channels); right now they are impossible to delete once added and they might leak memory although I think we do ultimately delete them when the session is closed.

Open to discussion on how to improve the UX around this.

azonenberg, 2023-11-20:
For starters, what would you expect to happen if you try to delete a node which is still in use?

Should it be deleted immediately without prompting, removing it from all waveform display and disconnecting any inputs that it was connected to?

Should it pop up a confirmation message if other filters are using its output?

Should the deletion silently be refused?

azonenberg, 2023-11-20:
The same reference counting also applies to instrument inputs: a channel which has at least one consumer is considered active (data will be downloaded from the scope and the channel is enabled in hardware). Once it reaches zero references it will be disabled (which may allow higher sample rates or more memory depth on some scopes) and its node will disappear from the filter graph.

MatFi, 2024-12-22:
> As of now, filter objects and channels are reference counted and automatically deleted when the last consumer (measurement display, sink / export filter, waveform view, or filter/channel input) is removed. So we don't allow deletion of a node explicitly, you simply stop using it and it goes away.

Thank you so much! I also was looking for men/button/key to delete nodes. This has really helped clear up the  confusion I had. If there's any way at all to see where the node is referenced, that would be great (may be in form of a popup?) .

azonenberg, 2024-12-22:
If we had that data, that would be 90% of the way to actually being able to properly implement deletion. Right now all we have is a reference count, and even that is sometimes buggy (occasionally channels continue to exist even when every reference I can find has been removed).

That whole lifetime tracking flow is problematic and needs to be ripped out and replaced by something better, as well as forward and reverse records of who is using a node. Once that's done, delete-on-request will become straightforward.

eqvinox, 2026-01-12:
> Sometimes? You don't. There were known bugs that could cause a reference to be leaked, I've fixed one since v0.1.1 but others probably exist. Saving and reopening the session sometimes fixes it.

Indeed, it appears to randomly work sometimes and sometimes not. Unfortunately reopening the session isn't a good workaround since the SDS1↊04X-E (all I have :sad panda:) has some issues with initializing the scope (particularly trigger conditions and Y scale) on load-up. I don't have resources to put into this (and I'm not a C++ dev sadly), and it's usable, so… I'll take it. Thanks for the awesome software! 😁