ngscopeclient/scopehal-apps #674: macOS build fails:  glslang target was not found

sslupsky, 2024-01-25:
I am attempting to build ngscopesclient on macOS Sonoma 14.2.1 but running into a Cmake error.  Build appears to correctly identify the location of the Vulkan SDK but fails to find glslang for some reason.

```
(base) stevenslupsky@Ducky build % which glslang
/Users/stevenslupsky/VulkanSDK/1.3.275.0/macOS/bin/glslang
```
[image]

```
(base) stevenslupsky@Ducky build % cmake ../ -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr \
             -DCMAKE_PREFIX_PATH="/opt/homebrew;/opt/homebrew/opt/libomp"

-- The C compiler identification is AppleClang 15.0.0.15000100
-- The CXX compiler identification is AppleClang 15.0.0.15000100
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.39.3 (Apple Git-145)") 
-- Found PkgConfig: /opt/homebrew/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'cairomm-1.0'
--   Found cairomm-1.0, version 1.14.5
-- Checking for module 'sigc++-2.0'
--   Found sigc++-2.0, version 2.10.8
-- Found OpenMP_C: -Xclang -fopenmp (found version "5.0") 
-- Found OpenMP_CXX: -Xclang -fopenmp (found version "5.0") 
-- Found OpenMP: TRUE (found version "5.0")  
-- Found Vulkan: /Users/stevenslupsky/VulkanSDK/1.3.275.0/macOS/lib/libvulkan.dylib (found version "1.3.275") found components: shaderc_combined glslc glslangValidator missing components: glslang
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found YAML: /opt/homebrew/lib/libyaml-cpp.dylib  
-- Found components for YAML
-- YAML_ROOT_DIR  = /usr
-- YAML_INCLUDES  = /opt/homebrew/include
-- YAML_LIBRARIES = /opt/homebrew/lib/libyaml-cpp.dylib
-- Found Vulkan: /Users/stevenslupsky/VulkanSDK/1.3.275.0/macOS/lib/libvulkan.dylib (found version "1.3.275") found components: glslc glslangValidator 
-- Found ZLIB: /Library/Developer/CommandLineTools/SDKs/MacOSX14.2.sdk/usr/lib/libz.tbd (found version "1.2.12")  
-- Found PNG: /opt/homebrew/lib/libpng.dylib (found version "1.6.40") 
Skipping MSI package build; define WIXPATH to enable
CMake Deprecation Warning at src/nativefiledialog-extended/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

nfd Platform: PLATFORM_MACOS
nfd Compiler: COMPILER_GNU
-- Performing Test NFD_USE_ALLOWEDCONTENTTYPES
-- Performing Test NFD_USE_ALLOWEDCONTENTTYPES - Success
-- Configuring done (1.9s)
CMake Error at lib/scopehal/CMakeLists.txt:244 (target_link_libraries):
  Target "scopehal" links to:

    Vulkan::glslang

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

-- Generating done (0.1s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

sslupsky, 2024-01-26:
Thank you for the feedback.

\> EDIT: are you using the standalone Vulkan SDK? I've been doing a major CMake rework that should resolve many of these problems.

Yes, I downloaded the latest Vulkan SDK (1.3.275.0) from the LunarG website and installed it.  When the brew formula PR is merged, should I install glslang using brew?

sslupsky, 2024-01-26:
One thing I noticed that stood out to me,
```
found components: shaderc_combined glslc glslangValidator missing components: glslang
```

glslangValidator is an alias of glslang.  So, if it found glslangValidator, how could it not find glslang?

d235j, 2024-01-27:
@sslupsky that is an issue in `FindVulkan.cmake` which I've worked around in my PR. 

> When the brew formula PR is merged, should I install glslang using brew?

With my CMake PR (#673), you will be able to use Vulkan SDK components either from brew or from the Vulkan SDK. I'm working on updating docs as well (https://github.com/ngscopeclient/scopehal-docs/pull/77).

sslupsky, 2024-04-12:
Looks like it the Homebrew fix was merged on Mar 6.

azonenberg, 2025-06-24:
Going through old tickets, I believe all of the problems encountered here have been addressed so we should be able to close this.

@sslupsky can you confirm?

azonenberg, 2025-06-29:
I think this is now fixed, please reopen if you are still having build issues.

closed 2025-06-29 by azonenberg