ngscopeclient/scopehal-apps #696: Changing units causes histogram to either not display data or to crash

mx-shift, 2024-02-17:
Built at 58af58e on Fedora 39.  nVidia Quadro P600 with nVidia 545.29.06 drivers.

If I import [00.0.revolution1.500_flashfloppy_v341.csv](https://github.com/ngscopeclient/scopehal-apps/files/14317939/00.0.revolution1.500_flashfloppy_v341.csv), I get a waveform display with the Y units default to volts.  I can create a histogram and get data that appears to be correct.

The Y units are actually phase error measured in clock ticks so I tried to guess a more appropriate unit.  Choosing either UI or dimensionless before creating a histogram, the waveform renders fine but the histogram is blank:
[image]

If I change the Y units to UI or dimensionless after opening a histogram, I get a crash:
```
mxshift@eps-conduit .../build/src/ngscopeclient ❯ ./ngscopeclient
Floating point exception (core dumped)
```

mx-shift, 2024-02-17:
```
Thread 8 "WaveformThread" received signal SIGFPE, Arithmetic exception.
[Switching to Thread 0x7fffc3e006c0 (LWP 13113)]
0x0000000000945500 in WaveformArea::RasterizeAnalogOrDigitalWaveform (this=0x13dcfc0,
    channel=std::shared_ptr<DisplayedChannel> (use count 3, weak count 0) = {...}, cmdbuf=..., clearPersistence=true)
    at /home/mxshift/Projects/thirdparty/github/ngscopeclient/scopehal-apps/src/ngscopeclient/WaveformArea.cpp:1768
1768		int64_t offset_samples = (offset - data->m_triggerPhase) / data->m_timescale;
Missing separate debuginfos, use: dnf debuginfo-install xorg-x11-drv-nvidia-libs-545.29.06-2.fc39.x86_64
(gdb) bt
#0  0x0000000000945500 in WaveformArea::RasterizeAnalogOrDigitalWaveform
    (this=0x13dcfc0, channel=std::shared_ptr<DisplayedChannel> (use count 3, weak count 0) = {...}, cmdbuf=..., clearPersistence=true) at /home/mxshift/Projects/thirdparty/github/ngscopeclient/scopehal-apps/src/ngscopeclient/WaveformArea.cpp:1768
#1  0x00000000009465d9 in WaveformArea::RenderWaveformTextures
    (this=0x13dcfc0, cmdbuf=..., chans=std::vector of length 1, capacity 1 = {...}, clearPersistence=clearPersistence@entry=false) at /home/mxshift/Projects/thirdparty/github/ngscopeclient/scopehal-apps/src/ngscopeclient/WaveformArea.cpp:1709
#2  0x000000000095d314 in WaveformGroup::RenderWaveformTextures
    (this=<optimized out>, cmdbuf=..., channels=std::vector of length 1, capacity 1 = {...}, clearPersistence=clearPersistence@entry=false) at /home/mxshift/Projects/thirdparty/github/ngscopeclient/scopehal-apps/src/ngscopeclient/WaveformGroup.cpp:166
#3  0x000000000082b0c6 in MainWindow::RenderWaveformTextures
    (this=<optimized out>, cmdbuf=<optimized out>, channels=<optimized out>)
    at /home/mxshift/Projects/thirdparty/github/ngscopeclient/scopehal-apps/src/ngscopeclient/MainWindow.cpp:487
#4  0x00000000008c6aec in Session::RenderWaveformTextures
    (this=<optimized out>, cmdbuf=<optimized out>, channels=<optimized out>)
    at /home/mxshift/Projects/thirdparty/github/ngscopeclient/scopehal-apps/src/ngscopeclient/Session.cpp:3429
#5  0x0000000000969fd4 in RenderAllWaveforms
    (cmdbuf=..., session=0xfb2148, queue=std::shared_ptr<QueueHandle> (use count 3, weak count 0) = {...})
    at /home/mxshift/Projects/thirdparty/github/ngscopeclient/scopehal-apps/src/ngscopeclient/WaveformThread.cpp:167
#6  0x000000000096a844 in WaveformThread (session=0xfb2148, shuttingDown=0xfb22f0)
    at /home/mxshift/Projects/thirdparty/github/ngscopeclient/scopehal-apps/src/ngscopeclient/WaveformThread.cpp:109
#7  0x00007ffff4ee31e3 in std::execute_native_thread_routine (__p=0x38df6a0)
    at ../../../../../libstdc++-v3/src/c++11/thread.cc:104
#8  0x00007ffff4bcb897 in start_thread (arg=<optimized out>) at pthread_create.c:444
#9  0x00007ffff4c5280c in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
(gdb)
```

azonenberg, 2024-02-17:
This is suggestive of the histogram somehow ending up with X axis units of zero UI per pixel, leading to a divide-by-zero in the renderer.

Not immediately obvious how this is happening. Can you share the CSV?

mx-shift, 2024-02-17:
It's linked in https://github.com/ngscopeclient/scopehal-apps/issues/696#issue-2139856198

azonenberg, 2024-03-13:
Hey I finally had some time to dig into this but the test file you uploaded is giving github cache errors and I can't download it. Not sure if it's just my local server or what but can you re-upload the test case?