ngscopeclient/scopehal-apps #703: Stop using old LINUX CMake variable and replace with checks for CMAKE_SYSTEM_NAME

azonenberg, 2024-03-05:
Check all build scripts incl submodules and remove them, seems to cause problems on recent Mint among others.

alphafox02, 2024-03-16:
Anything I can do in the meantime with 22.04?

azonenberg, 2024-03-16:
@d235j is working on a proper fix.

I think the basic idea is to replace if(LINUX) with if(CMAKE_SYSTEM_NAME MATCHES "Linux") every time it occurs in the build scripts. If you want to try that, let us know how it worked.

alphafox02, 2024-03-16:
Thank you. I will try asap. Curious to see how the uhd bridge works and this application works with various uhd hardware.

azonenberg, 2024-03-16:
Great. I've only tested it on the AntSDR so far although I do have a B205 on loan from a friend that I keep meaning to find time to test it with.

Do note that there's not a whole lot of filter graph blocks for processing I/Q data yet, you can do FFTs and spectrograms and some minimal downconversion but we don't have much in the way of demodulation filters, although many are on the wishlist.

alphafox02, 2024-03-16:
I have a couple AntSDRs too, also 205, 210, and x310. Looking forward to trying, learning, and following this project

alphafox02, 2024-03-16:
I did not keep track of which CMakelists I changed, but I think it was maybe two or three that I did what you recommened. It built and runs now! I will open other tickets as I'm getting this currently with the X310 using the UHD Bridge. I'll grab a b205 mini instead.

```
[ERROR] [MULTI_USRP] Could not find radio on mboard 0 that matches subdev A:A
ERROR: UHD exception: LookupError: Could not find radio on mboard 0 that matches subdev A:A
Shutting down...
```

alphafox02, 2024-03-16:
> Great. I've only tested it on the AntSDR so far although I do have a B205 on loan from a friend that I keep meaning to find time to test it with.
>
> Do note that there's not a whole lot of filter [...]

B205 mini connects with duallan and 5025:5026. I need to go do a lot of reading to understand what to do from here, but again just letting you know it build with the suggested change you mentioned.

azonenberg, 2024-03-25:
@mandl @d235j is this all fixed now, can we close?

mandl, 2024-03-28:
Works for me with Ubuntu 22.04.4 LTS. We can close this case.

closed 2024-03-28 by azonenberg