ngscopeclient/scopehal-apps #708: Allow disabling of imgui "viewports"

smunaut, 2024-04-13:
So the "viewports" feature makes imgui dynamically create windows for when you want stuff to draw outside the main surface. Like tooltip and menu and popups and such, it will dynamically create a new window so you can have those go out of the main window area.

Problem is that when you have a tiling window manager that automatically arrange windows, that doesn't play nice because those suddenly get resized to full screen (or whatever area the tiling WM assigned).

Ideally they'd be created at "floating" windows but I couldn't find any WM hints or anything that worked properly so what I end up doing is just disable that viewports features by patching it out of the code.

Would be nice if this could be a preference thing.

azonenberg, 2024-04-13:
According to some folks on Mastodon the correct fix is

"tooltips and menus need override-redirect set to True on any WM."

I'm not sure how we'd tell ImGui that a top level window it created is only going to be used as a tooltip or menu? Keeping this ticket open as figuring that out is likely the better long term solution.

Lukas1811, 2026-02-28:
I ran into the same issue and had soccess in fixing it but I somehting feels weird there.
The simple fix is to set the preference for "Appearance/Windowing/Viewport Mode" to "Single window"

So under the hood the issue is `ImGui::GetIO().ConfigFlags |= ImGuiConfigFlags_ViewportsEnable;` in this part of [MainWindow.cpp](https://github.com/ngscopeclient/scopehal-apps/blob/7e2e8d4184ab116ded48b5b8be0c68189f4411ce/src/ngscopeclient/MainWindow.cpp#L183). I would assume that it's fine on non-tiling window managers.

What feels weird to me is that the tooltip only is created as a full window when hovering the button outside the icon to the right side (tested in the Filter Pallette window). The behavior is also different between the different filters (I assume it has to do with the icon itself). For me on i3wm the tooltip gets correctly created while hovering the icon of most filters directly.

Lukas1811, 2026-02-28:
I made a PR with a fix for the issue.
It actually had nothing to do with the icon as I thought before, it only had to to do with the length of the tooltip text.
The new window was created when the tooltip would exit the boundary of the main window, my fix confines the tooltip to the boundaries of the main window.