ngscopeclient/scopehal-apps #720: CSV import broken

VIPQualityPost, 2024-06-10:
I tried with the raw file from scope including headers, and the same data with the headers stripped out, neither worked. 
The number of samples does not seem detected properly (says 500kS) and the data loads in as a bunch of horizontal lines, with improper time per sample as well. 
[image]
[tek0001CH1.csv](https://github.com/user-attachments/files/15776301/tek0001CH1.csv)

VIPQualityPost, 2024-06-10:
Sorry, forgot to include an important detail : this is 1Mpts but I get the same behavior with 20Mpts, and 100kpts files.

azonenberg, 2024-08-25:
There's a couple of things going on here. The first is a parsing bug related to line 500001 where I don't know how to handle positive exponents (it thinks it's a header line for some reason). Should have a fix soon.