ngscopeclient/scopehal-apps #724: DPI scaling is incorrectly determined on X11 Linux

whitequark, 2024-06-23:
This happens because it uses GLFW and GLFW currently retrieves content scale via "folklore heuristics" (their words) which do not really work: https://github.com/glfw/glfw/blob/b35641f4a3c62aa86a0b3c983d163bc0fe36026d/src/x11_init.c#L990-L1023

On X11 there's no unified way to do this. The best you could do in a fully arch-independent manner is to query RandR display DPI, but that's kind of a pain to do. An alternative is to look at GDK/Qt environment variables.

whitequark, 2024-06-23:
I've submitted https://github.com/ngscopeclient/scopehal-apps/pull/725 as an example of how to make DPI scaling determination work pretty okay on X11 Linux, but it still needs to be tested on Wayland Linux, Windows, and macOS.

nemanjan00, 2024-07-01:
Scaling also seems to be completely broken on Wayland. 

I am not sure if cause is also GLFW

[image]
[image]

I have a hack (temporary "solution")  that might help some people. 

It does seem to at least fix cursor alignment to buttons and to render with reasonable resolution

```
gamescope -W 1440 -H 900 -r 60 -- ngscopeclient
```

[image]

MonarchActual, 2024-09-24:
I have the same issue on wayland (sway).
`gamescope` also does not work because of an issue with my intel integrated GPU.
As a workaround I use Xwayland using `XDG_SESSION_TYPE=x11`.
This works fine for me but I am using the default scale of 1 / DPI of 96.

whitequark, 2025-12-14:
I will have to test this, I'll do it at the same time as #1022.

whitequark, 2026-01-21:
Nope, ngscopeclient is still unusably tiny on KDE/X11 in revision 0dc295532fdb40fc84d3c0298b791ca121546209

whitequark, 2026-01-21:
The cause is likely that Xft.dpi is 96:

```
$ xrdb -query
*customization: -color
Xcursor.size:   48
Xcursor.theme:  breeze_cursors
Xft.antialias:  1
Xft.dpi:        96
Xft.hinting:    0
Xft.hintstyle:  hintnone
Xft.rgba:       none
```

I cannot set it to a larger value because then the scaling gets applied twice and instead of ngscopeclient being unusably tiny, every KDE (or Electron, I don't remember which of them broke) application becomes unusably large.