ngscopeclient/scopehal-apps #738: Scaling margin issues on Windows

VIPQualityPost, 2024-08-27:
When ngscopeclient launches, the window size is correct and the content is displayed properly. If you drag the window to become larger, the content scales, but some kind of sidebar or margin is calculated wrong and covers up the properly scaled content. This was a fresh build today on Win11. I think I remember some discussion about this margin calculation when we were looking at hidpi stuff a few months back.
[image]
[image]

VIPQualityPost, 2024-08-30:
OK, I just built on my personal computer (this issue was in reference to my work PC) and it's not happening. So I guess it's something related to the setup on that computer- the only thing I can think is that I am using a kind of weird/old video card, it was the only one supporting Vulkan that I could find (Quadro P4000).

sphereinabox, 2025-04-13:
I'm seeing the same. Just downloaded everything and did a fresh compile today on both a laptop and desktop.

- On my laptop (ryzen 7 6800U) when the window is resized larger, the rendered area doesn't increase
- On my desktop (radeon 7900xtx) when the window is resized larger, the rendered area does increase.

I initially thought this was due to the 200% dpi scaling, but even at 100% when at native res on my laptop I hit this.

Here's a screenshot, which also shows 200% DPI scaling

[image]