ngscopeclient/scopehal-apps #739: macOS build failing with ffts.h error

membranophonist, 2024-08-28:
I'm running into this error when building on macOS 14.16.1 with VulkanSDK 1.3.29.0:
```
cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_PREFIX_PATH="$(brew --prefix);$(brew --prefix)/opt/libomp"
CMake Warning (dev) at CMakeLists.txt:3 (enable_language):
  project() should be called prior to this enable_language() call.
This warning is for project developers.  Use -Wno-dev to suppress it.

Detected and using VULKAN_SDK at /Users/djwilson/VulkanSDK/1.3.290.0/macOS
CMake Error at CMakeLists.txt:138 (find_path):
  Could not find LIBFFTS_INCLUDE_DIRS using the following files: ffts.h

-- Configuring incomplete, errors occurred!
``` 

Any help would be appreciated!

VIPQualityPost, 2024-08-29:
Are you on intel or apple silicon?
I just did a fresh build a day or two ago on M1 and it worked fine. 
I do recall some issue about this from before: https://github.com/ngscopeclient/scopehal/issues/660#issuecomment-2052468509

azonenberg, 2024-08-29:
Yet another reason to finish the deprecation of FFTS sooner rather than later.

membranophonist, 2024-08-29:
> Are you on intel or apple silicon? I just did a fresh build a day or two ago on M1 and it worked fine. I do recall some issue about this from before: [ngscopeclient/scopehal#660 (comment)](https://github.com/ngscopeclient/scopehal/issues/660#issuecomment-2052468509)

x86.

azonenberg, 2024-08-30:
FFTS is gone from all non-test code as of 1a704437bf094bbcdfa214c1e5d1f0778fed149a, please pull latest and try again. (Note that docs aren't yet updated so it's still listed as a dependency for now)

membranophonist, 2024-09-01:
I pulled master and ran the build command again. I'm still getting an error related to ffts.h.

```
cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_PREFIX_PATH="$(brew --prefix);$(brew --prefix)/opt/libomp"
CMake Warning (dev) at CMakeLists.txt:3 (enable_language):
  project() should be called prior to this enable_language() call.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The CXX compiler identification is AppleClang 15.0.0.15000309
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The C compiler identification is AppleClang 15.0.0.15000309
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found glslc: /usr/local/bin/glslc
nfd Platform: PLATFORM_MACOS
nfd Compiler: COMPILER_GNU
CMake Error at CMakeLists.txt:246 (find_path):
  Could not find LIBFFTS_INCLUDE_DIRS using the following files: ffts.h

-- Configuring incomplete, errors occurred!
```

I tried commenting out lines 246 and 247 of CMakeLists.txt, but that just led to another error.

azonenberg, 2024-09-01:
Run CMake with -DBUILD_TESTING=OFF to disable building unit tests?

membranophonist, 2024-09-01:
With the -DBUILD_TESTING=OFF setting.

````
cmake .. -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Release -DCMAKE_PREFIX_PATH="$(brew --prefix);$(brew --prefix)/opt/libomp"
CMake Warning (dev) at CMakeLists.txt:3 (enable_language):
  project() should be called prior to this enable_language() call.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found glslc: /usr/local/bin/glslc
nfd Platform: PLATFORM_MACOS
nfd Compiler: COMPILER_GNU
-- Configuring done (0.1s)
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
/Users/djwilson/Applications/scopehal-apps/lib/scopehal/LIBFFTS_INCLUDE_DIRS
   used as include directory in directory /Users/djwilson/Applications/scopehal-apps/lib/scopehal

-- Generating done (0.3s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
````

azonenberg, 2024-09-01:
There's no reference to LIBFFTS_INCLUDE_DIRS in the scopehal CMakeLists anymore. Did you do a recursive pull including submodules?

membranophonist, 2024-09-01:
Thanks, @azonenberg! That seems to have done the trick. I guess I've always worked on projects with a single repo.

azonenberg, 2024-09-01:
Sounds good, closing as fixed.

closed 2024-09-01 by azonenberg