ngscopeclient/scopehal-apps #748: Copy DLLs needed to run ngscopeclient.exe as part of the build process

AleksaBjelogrlic, 2024-09-14:
After building ngscopeclient using the instructions from the manual, I am faced with a slew of DLL not found errors when running ngscopeclient.exe from outside the MSYS shell (as most users will want to do). Below is the list of DLLs that need to be copied into the build\src\ngscopeclient directory to get the exe to run standalone.

libgcc_s_seh-1.dll
libwinpthread-1.dll
libstdc++-6.dll
libgomp-1.dll
libsigc-2.0-0.dll
libpng16-16.dll
libcairomm-1.0-1.dll
zlib1.dll
libcairo-2.dll
libfontconfig-1.dll
libfreetype-6.dll
libpixman-1-0.dll
libbrotlidec.dll
libbz2-1.dll
libharfbuzz-0.dll
libexpat-1.dll
libbrotlicommon.dll
libglib-2.0-0.dll
libgraphite2.dll
libintl-8.dll
libpcre2-8-0.dll
libiconv-2.dll

fredzo, 2024-09-14:
You need to use the MSI installer to get all the DLLs copied along with the exe file.
See source code of documentation (published version is not up-to-date) for details on how to build the MSI installer: https://github.com/ngscopeclient/scopehal-docs/blob/32aa18b0b511e0a32a1d27e461dfec637a4fb264/section-ng-gettingstarted.tex#L337

AleksaBjelogrlic, 2024-09-15:
Thanks @fredzo, good to see that it'll be in the docs soon

bvernoux, 2024-11-21:
This is fixed in https://github.com/ngscopeclient/scopehal-apps/issues/569 see also https://github.com/ngscopeclient/scopehal-apps/issues/743

azonenberg, 2025-07-11:
Testing the MSI on a Windows system works correctly, if you're running a self-compiled version you'll need to run from the msys shell unless you build an install from it.

Closing this.

closed 2025-07-11 by azonenberg