ngscopeclient/scopehal-apps #757: Segmentation fault on CSV export

VIPQualityPost, 2024-09-24:
Fresh build, get a hard crash segfault when trying to export CSV. 
When I click the little `...` to select the output file, or selecting the `Export` button (both from modal dialog) causes the crash. Minimum use case is just importing a CSV file then connecting to CSV export directly via filter graph.

azonenberg, 2024-10-29:
Unable to reproduce with latest. Is this still happening for you? We fixed a bug in the file browser a while back (a9c8bc1a76eca530f6c47a059433185136d3e8d0) that may have been contributing to this.

What platform are you running on?

VIPQualityPost, 2024-10-30:
I was having this crash on both mac and windows, but I just tried on a fresh build and I also can't reproduce now, so it must have gotten fixed in your file browser work. Thanks for checking in!

closed 2024-10-30 by VIPQualityPost