ngscopeclient/scopehal-apps #774: Floating windows  occasionally get "stuck" and are unable to be closed

lethalbit, 2024-10-10:
This is annoying to reproduce, but i've hit it on multiple occasions, in this case with the Waveform View panel.

When the panel is undocked, either intentionally or accidentally when moving it around, it will float onto the desktop as a second window, once there it will occasionally get stuck,  as in you can't move the window, nor close it anymore, you can however, resize it.

I'm unsure the exact cause, and sometimes by playing with the window enough you can get it unstuck, but I've not found a fully reproducible method. It could be related to #651? Seems like that would cause some of the weirdness i'm seeing.

This is as of fd3974e, and it's happened before then too.

azonenberg, 2026-01-11:
@lethalbit is this still a problem? Can you provide detailed reproduction steps against git head if so?

Elemecca, 2026-01-12:
> So, I recently switched to Wayland from X11, and the `Multi Window` viewport mode is entirely busted, so it's acting as if it was always set to `Single Window`, which means I can't actually repro it.

ImGui doesn't currently support multi-viewport mode on Wayland, so you'll always be stuck with a single parent window, though you should still be able to move the dockable panels around inside it.

Assuming your environment supports XWayland, you can force ngscopeclient to use X11 by setting the environment variable `XDG_SESSION_TYPE=x11`, which should allow you to use multiple windows.