ngscopeclient/scopehal-apps #790: Rigol DS1054Z Socket read failed

krekr, 2024-10-28:
I'm running ngscopeclient 0.1-dev+c6a7d4d3 on a clean Ubuntu 24.04.1 LTS install, which works fine with the 'demo' oscilloscope. When I try to connect it to my Rigol scope, it says: `Warning: Socket read failed (errno=11, Resource temporarily unavailable)`. 

I can tell the program interacts with the scope, which periodically says `Invalid input!` on screen. The app will also sometimes freezes so long the Ubuntu taskmanager intervenes, showing `ngscopeclient 0.1-dev+c6a7d4d3 is not respond...`

Also, when trying to set the sample rate, this is printed to the terminal: `ERROR: RigolOscilloscope::GetInterleaveConflicts not implemented for this model`

From the manual (and various posts on social media) I got the impression that the Rigol DS1000-range was considered stable. Am I missing something or is this a bug?

azonenberg, 2024-10-28:
The GetInterleaveConflicts message probably means that the driver doesn't know what scope model it's talking to and is completely confused.

How are you connecting to the scope (USB, Ethernet, etc)? What's the exact connection path you're providing?

Try running ngscopeclient with ` --debug --trace SCPISocketTransport` (assuming you're connecting via Ethernet) and upload the resulting console output.

krekr, 2024-10-28:
It's connected over Ethernet: the path is 192.168.1.17:5555. Transport is `lan`.

```krekr@ubuntu:~/scopehal-apps/build/src/ngscopeclient$ ./ngscopeclient --debug --trace SCPISocketTransport
OMP_WAIT_POLICY not set to PASSIVE
Re-exec'ing with correct environment
Initializing Vulkan
    VK_KHR_get_physical_device_properties2: supported
    VK_KHR_xcb_surface: supported
    VK_KHR_xlib_surface: supported
    VK_EXT_debug_utils: supported
    Loader/API support available for Vulkan 1.3
    Vulkan 1.2 support available, requesting it
    Initializing glfw 3.3.10 X11 GLX EGL OSMesa clock_gettime evdev shared
    GLFW required extensions:
        VK_KHR_surface
        VK_KHR_xcb_surface
    Physical devices:
        Device 0: llvmpipe (LLVM 17.0.6, 128 bits)
            API version:            0x00403112 (0.1.3.274)
            Driver version:         0x00000001 (0.0.0.1)
            Vendor ID:              10005
            Device ID:              0000
            Device type:            CPU
            int64:                  yes
            int16:                  yes (allowed in SSBOs)
            int8:                   yes (allowed in SSBOs)
            Max image dim 2D:       16384
            Max storage buf range:  128 MB
            Max mem objects:        4294967295
            Max compute shared mem: 32 KB
            Max compute grp count:  65535 x 65535 x 65535
            Max compute invocs:     1024
            Max compute grp size:   1024 x 1024 x 1024
            Memory types:
                Type 0
                    Heap index: 0
                    Device local
                    Host visible
                    Host coherent
                    Host cached
            Memory heaps:
                Heap 0
                    Size: 3 GB
                    Device local
        Selected device 0
            Queue families (1 total)
                Queue type 0
                    Queue count:          1
                    Timestamp valid bits: 64
                    Graphics
                    Compute
                    Transfer
            Driver: vk::DriverId::eMesaLlvmpipe
            Enabling 64-bit float support
            Enabling 64-bit integer support
            Enabling 16-bit integer support
            Enabling 16-bit integer support for SSBOs
            Enabling 8-bit integer support
            Enabling 8-bit integer support for SSBOs
            Device has VK_KHR_push_descriptor, requesting it
            Device has VK_KHR_shader_non_semantic_info, requesting it
            Device has VK_EXT_memory_budget, requesting it
            Device has VK_EXT_shader_atomic_float, requesting it
            Using heap 0, type 0 for pinned host memory
            Using heap 0, type 0 for card-local memory
            Unified memory GPU optimizations are enabled
            Sorted queues:
                Family=0 Index=0 Flags=00000007
            QueueManager creating family=0 index=0 name=g_vkTransferQueue
    
[... 99 lines cut ...]
[SCPISocketTransport::SendCommand] [192.168.1.17] Sending :TRIG:STAT?
[SCPISocketTransport::SendCommand] [192.168.1.17] Sending :TRIG:STAT?
[SCPISocketTransport::SendCommand] [192.168.1.17] Sending :TRIG:STAT?
[SCPISocketTransport::SendCommand] [192.168.1.17] Sending :TRIG:STAT?
[SCPISocketTransport::SendCommand] [192.168.1.17] Sending :SING
[SCPISocketTransport::SendCommand] [192.168.1.17] Sending *WAI
[SCPISocketTransport::SendCommand] [192.168.1.17] Sending :TRIG:STAT?
Warning: Socket read failed (errno=11, Resource temporarily unavailable)
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:SOUR CHAN1
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:PRE?
    Warning: Socket read failed (errno=11, Resource temporarily unavailable)
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:STAR 1
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:STOP 250000
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:DATA?
    Warning: Socket read failed (errno=11, Resource temporarily unavailable)
    Warning: Socket read failed (errno=11, Resource temporarily unavailable)
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:STAR 250001
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:STOP 500000
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:DATA?
    Warning: Socket read failed (errno=11, Resource temporarily unavailable)
    Warning: Socket read failed (errno=11, Resource temporarily unavailable)
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:STAR 500001
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:STOP 750000
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:DATA?
```

azonenberg, 2024-10-28:
So the driver is sending a bunch of commands to the scope and it's getting no replies whatesoever. This isn't looking like a problem on the ngscopeclient side so far.

Have you tried rebooting the scope? If you netcat to the SCPI port on the scope and send `*IDN?` and a newline, do you get anything back?

krekr, 2024-10-28:
Thanks for the quick replies, that sounds reasonable. I'll try tomorrow!

azonenberg, 2024-10-28:
Actually, that may not be correct. It looks like we may actually have a bug in logging of command responses due to some recent refactoring. Give me a bit to poke at this on my end.

azonenberg, 2024-10-28:
Ok confirmed, there was a logging bug. Please update to f4ad90df3 (make sure to pull submodules recursively) and repeat the trace.

krekr, 2024-10-29:
It's building now. Oh - forgot to note it's running inside a Virtualbox VM (win 10 host) and there are several warnings during compilation, e.g.;
```
/home/krekr/scopehal-apps/lib/scopehal/FlowGraphNode.cpp: In member function ‘virtual void FlowGraphNode::Refresh(vk::raii::CommandBuffer&, std::shared_ptr<QueueHandle>)’:
/home/krekr/scopehal-apps/lib/scopehal/FlowGraphNode.cpp:98:16: warning: ‘virtual void FlowGraphNode::Refresh()’ is deprecated [-Wdeprecated-declarations]
   98 |         Refresh();
      |         ~~~~~~~^~
/home/krekr/scopehal-apps/lib/scopehal/FlowGraphNode.cpp:87:6: note: declared here
   87 | void FlowGraphNode::Refresh()
      |      ^~~~~~~~~~~~~
```

azonenberg, 2024-10-29:
Yep, that's a reminder to us that some legacy functions haven't been fully removed. It's expected and harmless, we're gonna refactor the rest of it out at some point but likely not in time for the v0.1 release EOY.

krekr, 2024-10-29:
I'm not too familiar with git so I just pulled the latest version (af3f7267), assuming the same change is in there. It seems to startup faster. Now I'm getting;
```
krekr@ubuntu:~/scopehal-apps/build/src/ngscopeclient$ ./ngscopeclient --debug --trace SCPISocketTransport
OMP_WAIT_POLICY not set to PASSIVE
Re-exec'ing with correct environment
Initializing Vulkan
    VK_KHR_get_physical_device_properties2: supported
    VK_KHR_xcb_surface: supported
    VK_KHR_xlib_surface: supported
    VK_EXT_debug_utils: supported
    Loader/API support available for Vulkan 1.3
    Vulkan 1.2 support available, requesting it
    Initializing glfw 3.3.10 X11 GLX EGL OSMesa clock_gettime evdev shared
    GLFW required extensions:
        VK_KHR_surface
        VK_KHR_xcb_surface
    Physical devices:
        Device 0: llvmpipe (LLVM 17.0.6, 128 bits)
            API version:            0x00403112 (0.1.3.274)
            Driver version:         0x00000001 (0.0.0.1)
            Vendor ID:              10005
            Device ID:              0000
            Device type:            CPU
            int64:                  yes
            int16:                  yes (allowed in SSBOs)
            int8:                   yes (allowed in SSBOs)
            Max image dim 2D:       16384
            Max storage buf range:  128 MB
            Max mem objects:        4294967295
            Max compute shared mem: 32 KB
            Max compute grp count:  65535 x 65535 x 65535
            Max compute invocs:     1024
            Max compute grp size:   1024 x 1024 x 1024
            Memory types:
                Type 0
                    Heap index: 0
                    Device local
                    Host visible
                    Host coherent
                    Host cached
            Memory heaps:
                Heap 0
                    Size: 3 GB
                    Device local
        Selected device 0
            Queue families (1 total)
                Queue type 0
                    Queue count:          1
                    Timestamp valid bits: 64
                    Graphics
                    Compute
                    Transfer
            Driver: vk::DriverId::eMesaLlvmpipe
            Enabling 64-bit float support
            Enabling 64-bit integer support
            Enabling 16-bit integer support
            Enabling 16-bit integer support for SSBOs
            Enabling 8-bit integer support
            Enabling 8-bit integer support for SSBOs
            Device has VK_KHR_push_descriptor, requesting it
            Device has VK_KHR_shader_non_semantic_info, requesting it
            Device has VK_EXT_memory_budget, requesting it
            Device has VK_EXT_shader_atomic_float, requesting it
            Using heap 0, type 0 for pinned host memory
            Using heap 0, type 0 for card-local memory
            Unified memory GPU optimizations are enabled
            Sorted queues:
                Family=0 Index=0 Flags=00000007
            QueueManager creating family=0 index=0 name=g_vkTransferQueue
    
[... 273 lines cut ...]
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:DATA?
[SCPISocketTransport::ReadRawData]     Got 2 bytes
    Warning: Socket read failed (errno=11, Resource temporarily unavailable)
[SCPISocketTransport::ReadRawData]     Failed to get 12 bytes (@ pos 0)
    Warning: Socket read failed (errno=11, Resource temporarily unavailable)
[SCPISocketTransport::ReadRawData]     Failed to get 250001 bytes (@ pos 0)
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:STAR 5500001
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:STOP 5750000
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:DATA?
[SCPISocketTransport::ReadRawData]     Got 2 bytes
    Warning: Socket read failed (errno=11, Resource temporarily unavailable)
[SCPISocketTransport::ReadRawData]     Failed to get 12 bytes (@ pos 0)
    Warning: Socket read failed (errno=11, Resource temporarily unavailable)
[SCPISocketTransport::ReadRawData]     Failed to get 250001 bytes (@ pos 0)
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:STAR 5750001
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:STOP 6000000
[SCPISocketTransport::SendCommand]     [192.168.1.17] Sending WAV:DATA?
[SCPISocketTransport::ReadRawData]     Got 2 bytes
    Warning: Socket read failed (errno=11, Resource temporarily unavailable)
[SCPISocketTransport::ReadRawData]     Failed to get 12 bytes (@ pos 0)
Killed

```

Which doesn't seem all that different (still no scope image). It's also hanging again sometimes.

krekr, 2024-10-29:
Ok, I'm a step further. I thought I'd update the 'scope firmware since that was a while ago. It was running `00.04.02.SP4`, now the latest available version `00.04.05.02.xx`. I didn't get the `Warning: Socket read failed (errno=11, Resource temporarily unavailable)`, but also no waveform. This might be because the scope doesn't show anything until it's recalibrated after the update (which it is doing now).
The screenshot programs now also work. I'll report back after the calibration. Hope I didn't waste your time because of this.

azonenberg, 2024-10-29:
If the old scope firmware didn't work, that's still valuable information. We can add a note to the documentation and maybe a warning in the driver if we detect the scope is running the buggy version.

krekr, 2024-11-06:
Just wanted to let you know it works now. So this seems tied to firmware version indeed. Unfortunately I have no idea which fw versions are buggy and which are not.
I hadn't realised how slow 0.1WFM/s is, haha. I'll have to figure out a way to decrease the sample rate/depth; the links for that don't seem to work for Rigol scopes?
At least now I feel confident about getting this to compile and run to consider getting a Thunderscope; I've been looking into ngscopeclient since a few years ago...

closed 2024-11-08 by krekr