ngscopeclient/scopehal-apps #798: Migrate from GLFW to SDL

azonenberg, 2024-11-02:
GLFW has a bunch of bugs related to mouse handling on Linux causing false mouse-up events when a new platform window is created.

There may be ugly hacks we can do on the backend, but SDL doesn't have this issue so that's probably the best path forward. We only make 20ish GLFW API calls so the port shouldn't be too difficult.

See: https://github.com/ocornut/imgui/issues/7922

Johnsel, 2024-11-03:
Hate to say it (actually I kinda enjoy it, sorry 🤣), but didn't I propose also switching to SDL2 already when we switched to imgui? It's a much newer and better backend than glfw.

But glad you're finally convinced. It'd open up some more interesting ports as well.

azonenberg, 2024-11-03:
Tagging just for posterity: it appears https://github.com/ocornut/imgui/issues/7733 (a minor annoyance that had been plaguing us for a while) is rooted in the same bug and can also be fixed by leaving GLFW.

ocornut, 2024-11-03:
FYI if there are hacks to do on the GLFW backend its worth adding them imho. All backends have hacks/quirks, as the windowing emulation of multi-viewports is generally a bit out of the typical expected scope of SDL/GLFW. I would just need to understand exactly what’s the low level issue (perhaps that info is already detailed in existing high-level issues).

azonenberg, 2024-11-03:
> FYI if there are hacks to do on the GLFW backend its worth adding them imho. All backends have hacks/quirks, as the windowing emulation of multi-viewports is generally a bit out of the typical [...]

It appears the root cause is https://github.com/glfw/glfw/issues/1562#issuecomment-827366508.

If you can add a workaround on the ImGui side, that would be great.

ocornut, 2024-11-04:
I posted a tentative workaround here:
https://github.com/ocornut/imgui/issues/7733#issuecomment-2454518221
Would be nice if you can also test it on your end.

azonenberg, 2024-11-05:
Keeping this open as a potential future trajectory for the project, but no longer a v0.2 priority now that there is an imgui workaround for the specific GLFW issues in play.