ngscopeclient/scopehal-apps #800: Compile fails on Windows (Could not find a package configuration file provided by "yaml-cpp")

cdwijs, 2024-11-03:
I'm following this guide: https://www.ngscopeclient.org/manual/GettingStarted.html section 3.3 (windows), 7 (Build manually):

Submodule path 'src/imgui-node-editor': checked out '520db2c29a32364cd920f4966ad9ac1e3529c2c9'
Submodule path 'src/imgui_markdown': checked out '61a181bdb83f450f852f7cf5d1282d8cda1c0f57'
Submodule path 'src/nativefiledialog-extended': checked out '388549a5badaa7cbd138f5f189f50c67d5bf060c'

cedric@DESKTOP-LRFL83R UCRT64 ~
$ cd scopehal-apps

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps
$ mkdir build

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps
$ cd build

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps/build
$ cmake ..
-- Building for: Ninja
CMake Warning (dev) at CMakeLists.txt:3 (enable_language):
  project() should be called prior to this enable_language() call.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The CXX compiler identification is GNU 14.1.0
CMake Warning (dev) at C:/msys64/ucrt64/share/cmake/Modules/Platform/Windows-GNU.cmake:193 (enable_language):
  project() should be called prior to this enable_language() call.
Call Stack (most recent call first):
  C:/msys64/ucrt64/share/cmake/Modules/Platform/Windows-GNU-CXX.cmake:2 (__windows_compiler_gnu)
  C:/msys64/ucrt64/share/cmake/Modules/CMakeCXXInformation.cmake:48 (include)
  CMakeLists.txt:3 (enable_language)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/msys64/ucrt64/bin/c++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The C compiler identification is GNU 14.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/ucrt64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/msys64/usr/bin/git.exe (found version "2.45.2")
-- Found PkgConfig: C:/msys64/ucrt64/bin/pkg-config.exe (found version "2.2.0")
CMake Error at CMakeLists.txt:109 (find_package):
  By not providing "Findyaml-cpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "yaml-cpp",
  but CMake did not find one.

  Could not find a package configuration file provided by "yaml-cpp" with any
  of the following names:

    yaml-cppConfig.cmake
    yaml-cpp-config.cmake

  Add the installation prefix of "yaml-cpp" to CMAKE_PREFIX_PATH or set
  "yaml-cpp_DIR" to a directory containing one of the above files.  If
  "yaml-cpp" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps/build

My system:
Windows 10 Pro 64 bit
Version	22H2
OS build	19045.5011
Experience	Windows Feature Experience Pack 1000.19060.1000.0

I used "msys2-x86_64-20240727(1).exe"

bvernoux, 2024-11-05:
You shall check up to date manual https://github.com/ngscopeclient/scopehal-docs/blob/master/section-ng-gettingstarted.tex#L293 which include dependencies to be installed with MSYS2 like yaml-cpp ...

cdwijs, 2024-11-09:
I've tried all the steps again, now I get a different error:

In detail:

-> Delete c:\msys64
-> install msys2-x86_64-20240727.exe
-> Open https://www.ngscopeclient.org/manual/GettingStarted.html
-> pacman -S git wget mingw-w64-ucrt-x86\_64-cmake mingw-w64-ucrt-x86\_64-toolchain
Install all:
Packages (96) Total Installed Size:  993.99 MiB

-> pacman -S mingw-w64-ucrt-x86\_64-libsigc++ mingw-w64-ucrt-x86\_64-cairomm mingw-w64-ucrt-x86\_64-yaml-cpp mingw-w64-ucrt-x86\_64-glfw mingw-w64-ucrt-x86\_64-catch
Packages (17) Total Installed Size:  76.73 MiB

-> pacman -S mingw-w64-ucrt-x86\_64-vulkan-headers mingw-w64-ucrt-x86\_64-vulkan-loader mingw-w64-ucrt-x86\_64-shaderc \ 
mingw-w64-ucrt-x86\_64-glslang mingw-w64-ucrt-x86\_64-spirv-tools

cedric@DESKTOP-LRFL83R UCRT64 ~
$ pacman -S mingw-w64-ucrt-x86\_64-vulkan-headers mingw-w64-ucrt-x86\_64-vulkan-loader mingw-w64-ucrt-x86\_64-shaderc \
mingw-w64-ucrt-x86\_64-glslang mingw-w64-ucrt-x86\_64-spirv-tools
error: target not found:
-bash: mingw-w64-ucrt-x86_64-glslang: command not found

-> remove the extra "\", and combine into 1 single line:
$ pacman -S mingw-w64-ucrt-x86\_64-vulkan-headers mingw-w64-ucrt-x86\_64-vulkan-loader mingw-w64-ucrt-x86\_64-shaderc mingw-w64-ucrt-x86\_64-glslang mingw-w64-ucrt-x86\_64-spirv-tools
resolving dependencies...
looking for conflicting packages...

Packages (5) mingw-w64-ucrt-x86_64-glslang-14.3.0-1  mingw-w64-ucrt-x86_64-shaderc-2024.1-1
             mingw-w64-ucrt-x86_64-spirv-tools-3~1.3.290.0-1
             mingw-w64-ucrt-x86_64-vulkan-headers-1.3.290.0-1
             mingw-w64-ucrt-x86_64-vulkan-loader-1.3.290.0-1

Total Download Size:    9.65 MiB
Total Installed Size:  90.13 MiB

-> cd ~ 
git clone --recursive https://github.com/ngscopeclient/scopehal-apps

-> cd scopehal-apps 
mkdir build 
cd build 
cmake .. 
ninja

cedric@DESKTOP-LRFL83R UCRT64 ~
$ cd scopehal-apps
mkdir build
cd build
cmake ..
ninja
-- Building for: Ninja
CMake Warning (dev) at CMakeLists.txt:3 (enable_language):
  project() should be called prior to this enable_language() call.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The CXX compiler identification is GNU 14.1.0
CMake Warning (dev) at C:/msys64/ucrt64/share/cmake/Modules/Platform/Windows-GNU.cmake:193 (enable_l
anguage):
  project() should be called prior to this enable_language() call.
Call Stack (most recent call first):
  C:/msys64/ucrt64/share/cmake/Modules/Platform/Windows-GNU-CXX.cmake:2 (__windows_compiler_gnu)
  C:/msys64/ucrt64/share/cmake/Modules/CMakeCXXInformation.cmake:48 (include)
  CMakeLists.txt:3 (enable_language)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/msys64/ucrt64/bin/c++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The C compiler identification is GNU 14.1.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/ucrt64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: C:/msys64/usr/bin/git.exe (found version "2.45.2")
-- Found PkgConfig: C:/msys64/ucrt64/bin/pkg-config.exe (found version "2.2.0")
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Found OpenMP_C: -fopenmp (found version "4.5")
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Found glslc: C:/msys64/ucrt64/bin/glslc.exe
CMake Error at lib/xptools/CMakeLists.txt:11 (find_package):
  By not providing "Findhidapi.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "hidapi", but
  CMake did not find one.

  Could not find a package configuration file provided by "hidapi" with any
  of the following names:

    hidapiConfig.cmake
    hidapi-config.cmake

  Add the installation prefix of "hidapi" to CMAKE_PREFIX_PATH or set
  "hidapi_DIR" to a directory containing one of the above files.  If "hidapi"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
ninja: error: loading 'build.ninja': The system cannot find the file specified.

cdwijs, 2024-11-09:
After installing hidapi, the build fails on another missing dependency:

$ pacman -S mingw-w64-ucrt-x86_64-hidapi
resolving dependencies...
looking for conflicting packages...

Packages (1) mingw-w64-ucrt-x86_64-hidapi-0.14.0-2

Total Download Size:   0.05 MiB
Total Installed Size:  0.18 MiB

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps/build
$ cmake ..
ninja
CMake Warning (dev) at CMakeLists.txt:3 (enable_language):
  project() should be called prior to this enable_language() call.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at C:/msys64/ucrt64/share/cmake/Modules/Platform/Windows-GNU.cmake:193 (enable_l
anguage):
  project() should be called prior to this enable_language() call.
Call Stack (most recent call first):
  C:/msys64/ucrt64/share/cmake/Modules/Platform/Windows-GNU-CXX.cmake:2 (__windows_compiler_gnu)
  C:/msys64/ucrt64/share/cmake/Modules/CMakeCXXInformation.cmake:48 (include)
  CMakeLists.txt:3 (enable_language)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found glslc: C:/msys64/ucrt64/bin/glslc.exe
-- Found ZLIB: C:/msys64/ucrt64/lib/libz.dll.a (found version "1.3.1")
-- Found PNG: C:/msys64/ucrt64/lib/libpng.dll.a (found version "1.6.43")
Skipping MSI package build; define WIXPATH to enable
nfd Platform: PLATFORM_WIN32
nfd Compiler: COMPILER_GNU
CMake Error at CMakeLists.txt:264 (find_path):
  Could not find LIBFFTS_INCLUDE_DIRS using the following files: ffts.h

-- Configuring incomplete, errors occurred!
ninja: error: loading 'build.ninja': The system cannot find the file specified.

cdwijs, 2024-11-09:
I find this a bit odd, as the string "hidapi" is not found on the webpage containing the entire get started guide:
https://www.ngscopeclient.org/manual/GettingStarted.html

So I find it odd that it's missing for every supported platform.  I guess I missed something else.

azonenberg, 2024-11-09:
Nope, that's on us. hidapi was a fairly recently added dependency (last month or so) and while the LaTeX doc source code may have been updated (I need to double check) I haven't pushed PDF/HTML renders since. Will do that later today.

azonenberg, 2024-11-09:
FFTS is not required to build if you're not building unit tests. Can you check the generated CMakeCache.txt and see if BUILD_TESTING is on or off? Set to off and it should build fine without FFTS. We plan to transition these to FFTW eventually (the GPL dependency is a non-issue since tests aren't redistributed) but haven't got around to it.

cdwijs, 2024-11-11:
too long did't read: you are right, fixed:

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps/build
$ grep BUILD CMakeCache.txt
BUILD_DEVDOCS:BOOL=
BUILD_DOCS:BOOL=
BUILD_SHARED_LIBS:BOOL=OFF
BUILD_TESTING:BOOL=ON
CMAKE_BUILD_TYPE:STRING=
NFD_BUILD_SDL2_TESTS:BOOL=OFF
NFD_BUILD_TESTS:BOOL=OFF

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps/build
$ cp CMakeCache.txt CMakeCache.txt.bu

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps/build
$ nano CMakeCache.txt

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps/build
$ grep BUILD CMakeCache.txt
BUILD_DEVDOCS:BOOL=
BUILD_DOCS:BOOL=
BUILD_SHARED_LIBS:BOOL=OFF
BUILD_TESTING:BOOL=OFF
CMAKE_BUILD_TYPE:STRING=
NFD_BUILD_SDL2_TESTS:BOOL=OFF
NFD_BUILD_TESTS:BOOL=OFF

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps/build
$ nano CMakeCache.txt

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps/build
$ make
-bash: make: command not found

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps/build
$ ninja
ninja: error: loading 'build.ninja': The system cannot find the file specified.

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps/build
$ cmake ..
CMake Warning (dev) at CMakeLists.txt:3 (enable_language):
  project() should be called prior to this enable_language() call.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at C:/msys64/ucrt64/share/cmake/Modules/Platform/Windows-GNU.cmake:193 (enable_language):
  project() should be called prior to this enable_language() call.
Call Stack (most recent call first):
  C:/msys64/ucrt64/share/cmake/Modules/Platform/Windows-GNU-CXX.cmake:2 (__windows_compiler_gnu)
  C:/msys64/ucrt64/share/cmake/Modules/CMakeCXXInformation.cmake:48 (include)
  CMakeLists.txt:3 (enable_language)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found glslc: C:/msys64/ucrt64/bin/glslc.exe
Skipping MSI package build; define WIXPATH to enable
nfd Platform: PLATFORM_WIN32
nfd Compiler: COMPILER_GNU
-- Configuring done (2.0s)
-- Generating done (0.3s)
-- Build files have been written to: C:/msys64/home/cedric/scopehal-apps/build

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps/build
$ grep BUILD CMakeCache.txt
BUILD_DEVDOCS:BOOL=
BUILD_DOCS:BOOL=
BUILD_SHARED_LIBS:BOOL=OFF
BUILD_TESTING:BOOL=OFF
CMAKE_BUILD_TYPE:STRING=
NFD_BUILD_SDL2_TESTS:BOOL=OFF
NFD_BUILD_TESTS:BOOL=OFF

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps/build
$ ninja
[85/499] Building CXX object lib/scopehal/CMakeFiles/scopehal.dir/Unit.cpp.obj
In file included from C:/msys64/ucrt64/include/yaml-cpp/parser.h:13,
                 from C:/msys64/ucrt64/include/yaml-cpp/yaml.h:10,
                 from C:/msys64/home/cedric/scopehal-apps/lib/scopehal/scopehal.h:60,
                 from C:/msys64/home/cedric/scopehal-apps/lib/scopehal/Unit.cpp:37:
C:/msys64/ucrt64/include/yaml-cpp/dll.h:22:65: note: '#pragma message: Defining YAML_CPP_API for DLL import'
   22 | #        pragma message( "Defining YAML_CPP_API for DLL import" )
      |                                                                 ^
[86/499] Building CXX object lib/scopeprotocols/CMakeFiles/scopeprotocols.dir/ACCoupleFilter.cpp.obj
In file included from C:/msys64/ucrt64/include/yaml-cpp/parser.h:13,
                 from C:/msys64/ucrt64/include/yaml-cpp/yaml.h:10,
                 from C:/msys64/home/cedric/scopehal-apps/lib/scopehal/scopehal.h:60,
                 from C:/msys64/home/cedric/scopehal-apps/lib/scopeprotocols/ACCoupleFilter.cpp:30:

                from C:/msys64/home/cedric/scopehal-apps/src/ngscopeclient/Session.cpp:35:
C:/msys64/ucrt64/include/yaml-cpp/dll.h:22:65: note: '#pragma message: Defining YAML_CPP_API for DLL import'
   22 | #        pragma message( "Defining YAML_CPP_API for DLL import" )
      |                                                                 ^
[499/499] Linking CXX executable src\ngscopeclient\ngscopeclient.exe

cedric@DESKTOP-LRFL83R UCRT64 ~/scopehal-apps/build

cdwijs, 2024-11-11:
How do I actually run it? When I navigate to, and double click the generated file I get these missing dll errors:
C:\msys64\home\cedric\scopehal-apps\build\src\ngscopeclient\ngscopeclient.exe
libgcc_s_seh-1.dll
libwinpthread-1.dll
libstdc++6.dll
libgomp-1.dll

cdwijs, 2024-11-11:
After copying these dll's from C:\msys64\ucrt64\bin
libgcc_s_seh-1.dll
libwinpthread-1.dll
libstdc++6.dll
libgomp-1.dll

I still miss these dll's
libpng16-16.dll
libsig-2.0-0.dll
libsigc-2.0-0.dll

cdwijs, 2024-11-11:
After also copying these dll's from C:\msys64\ucrt64\bin, the program runs:
libpng16-16.dll
libsigc-2.0-0.dll
zlib1.dll

[image]

bvernoux, 2024-11-11:
> After also copying these dll's from C:\msys64\ucrt64\bin, the program runs: libpng16-16.dll libsigc-2.0-0.dll zlib1.dll [...]

For the dependencies see a solution here
https://github.com/ngscopeclient/scopehal-apps/issues/743

cdwijs, 2024-11-14:
I would suggest adding a note in the documentation to manually copy that list of dll's, so the user has a working program after following the steps in the getting started guide.

Shall I submit a PR for that?

azonenberg, 2025-01-31:
Docs fix merged, closing

closed 2025-01-31 by azonenberg