ngscopeclient/scopehal-apps #802: Suggestion to make this tool epic (in my view)

Tobey16, 2024-11-07:
If it would have the features of the DSView, i would be really thrilled. Especially the multistage trigger, and the measurement features. And of course a timescale that has a triggerindication that starts at 0s.

I would even spend money on that.

azonenberg, 2024-11-07:
You're going to have to be more specific.

1) Trigger support is a function of the specific instrument firmware and our driver for it. Not all instruments support complex triggering in hardware, if they do the driver may or may not implement it.

If you want to request support for a specific, currently unsupported trigger type on a specific instrument, please file a separate ticket against ngscopeclient/scopehal.

2) What measurement features? We have a fairly large library of measurement blocks already, is there a specific measurement you want that we don't have? Please file separate tickets for each missing measurement.

3) Having T=0 as the start of the waveform, rather than the trigger timestamp, makes much more sense if you're decoupling the viewport position from the trigger position and not pretending to be an analog CRT scope.

Also, once you start adding multiple instruments to a single timeline, the T=0 reference becomes even less meaningful. If you have two or three scopes in a session, at most one could possibly be triggering at T=0 anyway. And once you start adding trigger path deskew, you have to account for the delta between the nominal and actual trigger position.

This was an explicit forward-looking design decision to help us overcome architectural limitations of legacy scope UIs and isn't going to change.

Tobey16, 2024-11-08:
1. Under supported hardware it says "DSLogic U3P16 via bridge server": Does it mean, that my DSLogic Plus can be used with its multi stage triggers?

2. Measurement features like this:
[image]

3. I think you didnt understood what i was meaning, because i dont understand what you are meaning ;-):
  At DSView, i have a trigger indication, that works as i excpect it (almost). Problem is, that trigger indication has a value of 32ms for example, but i need it to be 0ms, in order to do the right measurements on the traces (debugging). I would just need a timereference adjustment, if other considerations prevent changing it in another way.

azonenberg, 2024-11-13:
Snapping to data in various forms is also already planned. No ETA but see #763, #320.

Tobey16, 2024-11-13:
Multistage trigger:
Trigger when multiple conditions were met (in order):
- CH1 was high vor x us
- CH2 had x pulses
- CH3 has rising edge

azonenberg, 2024-11-13:
We can only implement triggers a given instrument provides, or do postprocessing on them.

If you have a specific target device in mind with such a trigger capability please file a feature request against scopehal. I know LeCroy has a (not yet supported) multistage trigger capability I need to add on the back end.