ngscopeclient/scopehal-apps #803: High load at idle

fgervais, 2024-11-09:
This is my first time with ngscopeclient so it might be something expected I'm not sure.

When ngscopeclient is running I see high load on the cpu and gpu.
Just the empty window doing nothing from the user perspective.

[image]

Is this an issue I should debug on my side or is it expected ?

I'm running scopehal-apps `195162a376d333a87ddf369d71e8695d08aed5f2`.

azonenberg, 2024-11-09:
This is normal and expected in the default mode with the event loop running as fast as it can. If you go to Setup | Preferences and expand Power / Events there's some tunables you can play with.

Event loop mode "performance" runs as fast as it can, while "power" blocks until the polling timeout expires (default 4 Hz / 250ms but configurable) or a mouse/keyboard event occurs. This provides a significant reduction in CPU/GPU usage but at the cost of limiting max waveform rate when live streaming from a scope.

At some point we want to try to do some better optimization to prevent needless redraws while still being able to grab scope waveforms as fast as they show up. But that probably won't happen until after v0.1.

(I don't think documentation has been updated to reflect this, but that will definitely happen before v0.1 is released)

azonenberg, 2024-11-09:
On closer inspection, though, this is still odd because it shouldn't be maxing every core. I see about 85% of a single core with it idle on my laptop in power mode. Keeping this ticket open for now.