ngscopeclient/scopehal-apps #804: Opening a *.scopesession with CSV data results in partial display

attie-argentum, 2024-11-11:
Opening the attached `*.scopesession` results in the presentation below. I'm not 100% certain if the decode is cached from before, or if it's genuine. Note, there appears to be (almost exactly?) 128ns of waveform at the left, and "a pixel or two" at the right.

The only way I've found to rectify this, is to re-open each CSV file, as depicted below.

[data.zip](https://github.com/user-attachments/files/17708577/data.zip)

[image]

[image]

[image]

attie-argentum, 2024-11-11:
Possibly related, I've also just managed to get multi-channel CSV working (it doesn't like more than on header row at all...)

It seems that some samples before `t=0` are missing.

[image]

azonenberg, 2024-11-12:
Investigating. Your screenshots certainly don't seem right.

The render looks fine for me, though. What OS are you using?

azonenberg, 2024-11-12:
This is what I get opening your file (after fixing paths, import filters use absolute paths now which have to be fixed when the session is moved to another computer).

[image]

It looks fine with no issues.

As far as decode output being cached, we don't cache any filter outputs in files. Only the configuration and the input waveforms are saved. So if the decode output is right, the data is correct internally and this is purely a display bug.

Also, what GPU and video driver version are you running? Is this native or in a VM?

attie-argentum, 2024-11-12:
> What OS are you using?

Windows 10, 64-bit (22H2, build 19045.5011)

> what GPU and video driver version are you running?

Nvidia RTX3080, v32.0.15.6094

> native or in a VM?

Running native, built in a VM (also Win 10, but I don't imagine that's a factor).

> import filters use absolute paths

Oops, I'll sanitize the archive and re-supply (I'll edit the first post).
I'll also place the files locally instead of on a network share, and see if that makes a difference.

attie-argentum, 2024-11-12:
There's no real change when opening from local disk, but a different channel is fully rendered...

[image]

attie-argentum, 2024-11-12:
I've attached the original CSV (but modified to only have a single header row), this has data back to -1.2μs.

[scope_20241111_165051.csv](https://github.com/user-attachments/files/17714189/scope_20241111_165051.csv)

azonenberg, 2024-11-12:
Hmm, weeeird. Nothing is jumping out at me as an obvious cause.

If you close and reopen a bunch of times, do you get the same or different results? This smells like a race condition to some extent (trying to render data that's not fully loaded or something) but I'm not yet sure how that would happen from looking at the code.

Any other Windows users, maybe @bvernoux or @fredzo , want to look at this test case and see if you can reproduce? Also would be very interested if anyone can make it happen on non-Windows or not.

azonenberg, 2024-11-12:
This bit in extended debug logs is interesting:

```
[CSVImportFilter::OnFileNameChanged]         Initial parsing completed, 9997 lines, 4 columns, 4 names, 9999 timestamps
[ImportFilter::TryNormalizeTimebase]         Min sample interval:     1000 ps
[ImportFilter::TryNormalizeTimebase]         Average sample interval: 879.763 ps
[ImportFilter::TryNormalizeTimebase]         Max sample interval:     18446.74407 s
[ImportFilter::TryNormalizeTimebase]         Stdev of intervals:      12.01120 ns
```

It thinks there are two samples 18000 seconds apart. If timestamps are getting corrupted somehow, that would absolutely explain the bad rendering since the shaders include a lot of optimizations assuming that sample timestamps are strictly monotonic (they don't have to be *uniform*, but sample X+1 must have a timestamp strictly greater than that of sample X).

azonenberg, 2024-11-12:
Ok so it looks like we have a bug where sample 0 in the CSV import filter has a timestamp of 0, then sample 1 has a valid timestamp.

This results in sample 0 having a negative duration, then a bunch of non-monotonic timestamps, then it recovers once the timebase hits zero. Still working on root-causing and fixing it, but this explains at least some of your problem.

azonenberg, 2024-11-12:
This should fix at least part of the issue. Can you update to 06e7130fdef8e40f1ce69e98332057feab05b559 (make sure to recursively pull submodules) and see if things work better now?

attie-argentum, 2024-11-12:
Not sure if you expected it to, but that appears to have resolved both issues! (data before `t=0` and multi-channel open on Windows)

Many thanks for such a quick fix!

closed 2024-11-12 by azonenberg

attie-argentum, 2024-11-12:
... to follow up, the missing data before `t=0` still occurs if there are multiple header rows... not sure if that's perhaps something you'd like to detect / handle / notify the user on.

azonenberg, 2024-11-12:
I'll file a separate issue for better handling of this in the CSV import filter.