ngscopeclient/scopehal-apps #824: Window scaling calculated improperly

VIPQualityPost, 2025-03-22:
Ubuntu 24.10 completely bone stock (GNOME / mutter). 
Rescaling the window does not rescale the contents by the appropriate size, including smashing text when adjusting window height.

Running with arguments `--debug --trace VulkanWindow` gives attached log.

[log.txt](https://github.com/user-attachments/files/19405053/log.txt)

[image]

https://github.com/user-attachments/assets/6f357acc-679a-411d-acc3-8298cb70adb0

azonenberg, 2025-03-22:
Looks like VulkanWindow::UpdateFramebuffer is not being called during resizes. This is probably a Wayland-specific bug, possibly a regression introduced by the recent work on resize handling.

azonenberg, 2025-04-23:
Possibly related? https://github.com/ocornut/imgui/issues/8587

VIPQualityPost, 2025-04-24:
Maybe, but I didn't have multiple mouse problem (or I need to double check, at least) and they say they couldn't resize or move window, but I could.

Ablu, 2025-07-14:
@azonenberg suggested that this it may be related to https://mastodon.social/@azonenberg@ioc.exchange/114852347369410095. But reverting that commit has no impact.

I toyed around a bit and this works for me: https://github.com/Ablu/scopehal-apps/commit/2e0d4ba6fcc58ff6f0a578b2d5fd77af5d94c622

Need to run now. Can try to build a proper solution later this week.