ngscopeclient/scopehal-apps #827: Windows build instructions fail at cmake stage

OllyFunkster, 2025-04-02:
Following instructions in msys ucrt64 shell on windows 10.

Listed dependencies install fine. When calling the default cmake (/ucrt64/bin/cmake) in the shell, nothing happens at all. I see some reports of similar behaviour out on the internet and they suggest explicitly calling /usr/bin/cmake.exe, which does at least do something but fails with:
```
$ /usr/bin/cmake.exe ..
System is unknown to cmake, create:
Platform/MINGW64_NT-10.0-19045 to use this system, please post your config file on discourse.cmake.org so it can be added to cmake
Your CMakeCache.txt file was copied to CopyOfCMakeCache.txt. Please post that file on discourse.cmake.org.
-- Found glslc: /ucrt64/bin/glslc.exe
-- Could NOT find ZLIB (missing: ZLIB_LIBRARY ZLIB_INCLUDE_DIR)
CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find PNG (missing: PNG_LIBRARY PNG_PNG_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake/Modules/FindPNG.cmake:177 (find_package_handle_standard_args)
  src/ngscopeclient/CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!

```

azonenberg, 2025-04-14:
@d235j any ideas? Haven't seen this one before and you've been doing windows testing a bunch

d235j, 2025-04-14:
`/usr/bin/cmake.exe` is the MSYS cmake and likely won't work.

> When calling the default cmake (/ucrt64/bin/cmake) in the shell, nothing happens at all.

This implies an issue with the MSYS2 environment. CI is currently working; I'll test with ucrt64 on my system but I would recommend starting out by updating (`pacman -Syu`, possibly a second time if it forces you to close the shell), and restarting the computer.

OllyFunkster, 2025-04-14:
Many thanks for the pointers - I had updated earlier, but somehow missed that there would be more updates needed if it forced you to close the shell. Updated and restarted the shell until it no longer had anything to do, and the default cmake now does something!

alas, I now have a differently error:
```
nfd Platform: PLATFORM_WIN32
nfd Compiler: COMPILER_GNU
-- Configuring done (11.0s)
CMake Error at lib/scopehal/CMakeLists.txt:240 (target_link_libraries):
  Target "scopehal" links to:

    glslang::SPIRV

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.
```

d235j, 2025-04-14:
Can you do a `git pull --recurse-submodules` and try again? I pushed some fixes for the above issue last night.

OllyFunkster, 2025-04-14:
Yeah that got it! Built and running! Thank you so much  🥰