ngscopeclient/scopehal-apps #830: MacOS libOMP not found with brew

VIPQualityPost, 2025-04-04:
This used to work, not sure what changed. CMake can't find libOMP on apple silicon clang.
I added this : 
```
if(CMAKE_C_COMPILER_ID MATCHES "Clang")
    set(OpenMP_C_FLAGS "-Xpreprocessor -fopenmp -I/opt/homebrew/opt/libomp/include")
    set(OpenMP_C_LIB_NAMES "omp")
    set(OpenMP_omp_LIBRARY "/opt/homebrew/opt/libomp/lib/libomp.dylib")
  endif()
  if(CMAKE_CXX_COMPILER_ID MATCHES "Clang")
    set(OpenMP_CXX_FLAGS "-Xpreprocessor -fopenmp -I/opt/homebrew/opt/libomp/include")
    set(OpenMP_CXX_LIB_NAMES "omp")
    set(OpenMP_omp_LIBRARY "/opt/homebrew/opt/libomp/lib/libomp.dylib")
  endif()
``` 
before searching for OpenMP and then the build is succeeding now. I got this snippet from here:  https://github.com/stella-cv/stella_vslam/issues/578#issuecomment-2673184480

azonenberg, 2025-04-14:
@lainy see above

azonenberg, 2025-06-24:
@lainy @d235j any ideas on what's going on here? This is a v0.1 release blocker and I can't reproduce on my mac mini

azonenberg, 2025-06-24:
Very possible. Are you using a rosetta setup? ngscopeclient should build and run fine under native aarch64.

d235j, 2025-06-24:
It sounds like you have remnants of x86_64 Homebrew in `/usr/local`, and the libomp and other Homebrew dependencies you have are x86_64, not arm64. arm64 Homebrew installs in `/opt/homebrew` not `/usr/local`.

 I recommend clearing this out, which likely means running the Homebrew uninstaller script in Rosetta and then manually cleaning out what's left there.

I do not recommend x86_64 Homebrew. If you need libraries for use under x86_64 or Rosetta, it's best to install in a separate prefix that isn't /usr/local.

VIPQualityPost, 2025-06-25:
Removed brew (both native and x86_64) using scripts here: https://github.com/homebrew/install?tab=readme-ov-file#uninstall-homebrew

However, I still get the issue! Even with logout/login and making sure to add env vars for libomp installations...

VIPQualityPost, 2025-06-29:
@d235j 
```
$: brew info libomp
==> libomp: stable 20.1.7 (bottled) [keg-only]
LLVM's OpenMP runtime library
https://openmp.llvm.org/
Installed
/opt/homebrew/Cellar/libomp/20.1.7 (9 files, 1.7MB)
  Poured from bottle using the formulae.brew.sh API on 2025-06-24 at 22:29:01
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/lib/libomp.rb
License: MIT
==> Dependencies
Build: cmake ✔, lit ✘
==> Caveats
libomp is keg-only, which means it was not symlinked into /opt/homebrew,
because it can override GCC headers and result in broken builds.

For compilers to find libomp you may need to set:
  export LDFLAGS="-L/opt/homebrew/opt/libomp/lib"
  export CPPFLAGS="-I/opt/homebrew/opt/libomp/include"
==> Analytics
install: 55,414 (30 days), 187,645 (90 days), 653,338 (365 days)
install-on-request: 15,926 (30 days), 48,685 (90 days), 148,722 (365 days)
build-error: 5 (30 days)
```
Trying to build: 
```
$: cmake ..
-- Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES) (found version "5.1")
-- Could NOT find OpenMP_CXX (missing: OpenMP_CXX_FLAGS OpenMP_CXX_LIB_NAMES) (found version "5.1")
-- Could NOT find OpenMP (missing: OpenMP_C_FOUND OpenMP_CXX_FOUND) 
CMake Error at CMakeLists.txt:155 (message):
  ngscopeclient requires OpenMP but your C++ compiler does not appear to
  support it.

-- Configuring incomplete, errors occurred!
```

Manually removed what was in /usr/local/lib (and /usr/local/bin) for libOMP :
```
$: ls /usr/local/lib | grep omp
libOGLCompiler.a
libdxcompiler.3.7.dylib
libdxcompiler.dylib
```

d235j, 2025-06-29:
You need to pass some parameters to `cmake` as is covered in section 3.3.2 of the manual at https://www.ngscopeclient.org/downloads/ngscopeclient-manual.pdf#page24 . Otherwise it won't be able to find OpenMP.

VIPQualityPost, 2025-06-29:
Doesn't matter, I get the same output: 
```
$: cmake .. -DCMAKE_BUILD_TYPE=Release -DCMAKE_PREFIX_PATH="$(brew --prefix);$(brew --\
prefix)/opt/libomp"
-- Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES) (found version "5.1")
-- Could NOT find OpenMP_CXX (missing: OpenMP_CXX_FLAGS OpenMP_CXX_LIB_NAMES) (found version "5.1")
-- Could NOT find OpenMP (missing: OpenMP_C_FOUND OpenMP_CXX_FOUND) 
CMake Error at CMakeLists.txt:155 (message):
  ngscopeclient requires OpenMP but your C++ compiler does not appear to
  support it.

-- Configuring incomplete, errors occurred!
```

d235j, 2025-06-29:
/opt/homebrew/Cellar/libomp/20.1.7 and /opt/homebrew/opt/libomp should both point to the same place. I’ll debug once I’m back at a Mac…

VIPQualityPost, 2025-12-10:
Bought a new machine, unable to repo, must have been something wonky in my old environment.