ngscopeclient/scopehal-apps #839: Delete key doesn't delete filters

darkstar007, 2025-05-31:
Hi,

Just downloaded and built "ngscopeclient 0.1-dev-9107f172". In the 'Filter Graph'  window I can't delete the filters with the delete key - I can delete the connections between filters - but the filters can't be deleted. No message appears to be printed anywehere.

There also seems to be no other way to delete them (I would, from a normal UI view, expect to be able to delete them from the RMB pop-up menu on a filter).

Thanks,
Matt

azonenberg, 2025-05-31:
Just to give a few more details on what will be involved for anybody curious (and to explain why this is such a big job):

- Filters are currently one of the few key objects that use raw new/delete rather than std::shared_ptr, this needs to change
- Filters and instrument channels are derived from the same base class, where the refcounting is implemented. When refcount hits zero, filters self-delete while instrument channels are turned off in hardware to avoid wasting bandwidth downloading waveform data nobody is going to use.
- The number of direct calls to AddRef/Release is fairly small (14/16 from some quick grepping), so at least fixing these won't be too bad.
- All code that handles instrument channel or filter objects will have to change from working with InstrumentChannel* to std::shared_ptr<InstrumentChannel>
- We need to make the filter graph data structure bidirectional (adding a list of sinks to each source) and make some new APIs to keep them in sync. These APIs will have to be breaking (changing variable names or making things private/protected) in order to enforce that all existing code is updated to use this new API and we never manipulate e.g. m_inputs directly
- We need to figure out how to turn off instrument channels when not in use given that we no longer have the Release() hook point
- We need to figure out how to manage instrument channels themselves since they will now have to be shared_ptr's and this will likely break some things drivers do

Once all of *this* is done, we can start thinking about using these new tools to implement the delete-filter feature everybody wants.

azonenberg, 2025-05-31:
(this refactoring is probably also going to be a good time to address https://github.com/ngscopeclient/scopehal/issues/970 which is also sorely needed but didn't have a ticket filed AFAIK until now)