ngscopeclient/scopehal-apps #855: Set up Fedora container for GH actions runner

azonenberg, 2025-06-30:
Doesn't seem like there is native support for hat-based distros but we can probably do something with docker?

Gherkin, 2025-07-09:
Got it building in fedora both with repo vulkan and standalone, also debian with repo, just have to verify with standalone.

I'm guessing that we don't care about building the docs? tested building them in the fedora container, but im having issues.

azonenberg, 2025-07-09:
Docs in the container are a lower priority, but it would be nice to have eventually. We'd like to be able to use the same workflow to create official binary package releases, and it will be easier to create packages for each distro in the appropriate container (meaning the doc package for fedora needs to build in the fedora container).

Gherkin, 2025-07-09:
not natively, but i guess it can be installed, they are not the most minimal containers. so that might be a possibility, from the arch issue linked in that https://github.com/ngscopeclient/scopehal-docs/issues/97 it seems like it needs a specific build flag, so that would also need to be included in the ctan-supplied version

azonenberg, 2025-07-09:
Arch would be nice to have building as well.

Which tests are failing? I've seen I think the de-embed test crashing intermittently on some github CI runners and working on every physical machine I've tested. Have not yet root caused it, but it's nothing to do with your build setup.

Gherkin, 2025-07-10:
Ill look into building Arch also :)

looking a bit at the test, it seems like it always is the Filter_EyePattern test that fails occasionally, weirdly enough by segfault. (here with repo)
```Test project /__w/scopehal-apps/scopehal-apps/build
      Start  1: Buffers_CpuOnly
 1/16 Test  #1: Buffers_CpuOnly ..................   Passed    0.07 sec
      Start  2: Buffers_CpuGpu
 2/16 Test  #2: Buffers_CpuGpu ...................   Passed    0.03 sec
      Start  3: Filter_Add
 3/16 Test  #3: Filter_Add .......................   Passed    0.10 sec
      Start  4: Filter_ACRMS
 4/16 Test  #4: Filter_ACRMS .....................   Passed    0.70 sec
      Start  5: Filter_DeEmbed
 5/16 Test  #5: Filter_DeEmbed ...................   Passed    2.00 sec
      Start  6: Filter_EyePattern
 6/16 Test  #6: Filter_EyePattern ................Subprocess aborted***Exception:   0.62 sec
Filters: "Filter_EyePattern"
Randomness seeded to: 3495703740
Baseline (expecting no mask hits)

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Filters is a Catch2 v3.8.0 host application.
Run with -? for options

-------------------------------------------------------------------------------
Filter_EyePattern
  Baseline
-------------------------------------------------------------------------------
/__w/scopehal-apps/scopehal-apps/tests/Filters/Filter_EyePattern.cpp:119
...............................................................................

/__w/scopehal-apps/scopehal-apps/tests/Filters/Filter_EyePattern.cpp:119: FAILED:
  {Unknown expression after the reported line}
due to a fatal error condition:
  SIGSEGV - Segmentation violation signal

double free or corruption (out)

      Start  7: Filter_FIR
 7/16 Test  #7: Filter_FIR .......................   Passed    0.34 sec
      Start  8: Filter_FFT
 8/16 Test  #8: Filter_FFT .......................   Passed    1.08 sec
      Start  9: Filter_Subtract
 9/16 Test  #9: Filter_Subtract ..................   Passed    0.12 sec
      Start 10: Filter_Upsample
10/16 Test #10: Filter_Upsample ..................   Passed    0.40 sec
      Start 11: Filter_FrequencyMeasurement
11/16 Test #11: Filter_FrequencyMeasurement ......   Passed    0.28 sec
      Start 12: Primitive_Averager
12/16 Test #12: Primitive_Averager ...............   Passed    0.52 sec
      Start 13: Primitive_Convert8BitSamples
13/16 Test #13: Primitive_Convert8BitSamples .....   Passed    0.14 sec
      Start 14: Primitive_Convert16BitSamples
14/16 Test #14: Primitive_Convert16BitSamples ....   Passed    0.16 sec
      Start 15: Primitive_FindZeroCrossings
15/16 Test #15: Primitive_FindZeroCrossings ......   Passed    1.72 sec
      Start 16: Primitive_SampleOnRisingEdges
16/16 Test #16: Primitive_SampleOnRisingEdges ....   Passed    0.18 sec

94% tests passed, 1 tests failed out of 16

Total Test time (real) =   8.48 sec

The following tests FAILED:
	  6 - Filter_EyePattern (Subprocess aborted)
Errors while running CTest
Error: Process completed with exit code 8.```

Also another weird issue. building fedora with standalone the tests refuse to run with the following error, everything builds fine though.

[... 66 lines cut ...]
16/16 Test #16: Primitive_SampleOnRisingEdges ....***Failed    0.01 sec
    ERROR: vk::SystemError: Context::createInstance: ErrorIncompatibleDriver

0% tests passed, 16 tests failed out of 16

Total Test time (real) =   0.20 sec

The following tests FAILED:
	  1 - Buffers_CpuOnly (Failed)
	  2 - Buffers_CpuGpu (Failed)
	  3 - Filter_Add (Failed)
	  4 - Filter_ACRMS (Failed)
	  5 - Filter_DeEmbed (Failed)
	  6 - Filter_EyePattern (Failed)
	  7 - Filter_FIR (Failed)
	  8 - Filter_FFT (Failed)
	  9 - Filter_Subtract (Failed)
	 10 - Filter_Upsample (Failed)
	 11 - Filter_FrequencyMeasurement (Failed)
	 12 - Primitive_Averager (Failed)
	 13 - Primitive_Convert8BitSamples (Failed)
	 14 - Primitive_Convert16BitSamples (Failed)
	 15 - Primitive_FindZeroCrossings (Failed)
	 16 - Primitive_SampleOnRisingEdges (Failed)
Errors while running CTest```

azonenberg, 2025-07-10:
Guessing it has to do with the vulkan setup in the VM. Maybe a mismatch of the vulkan headers and binary libs or something? Or complete lack of a vulkan implementation (normally these vms have llvmpipe in them if there's no GPU)?

I think I may have seen the eye pattern test segfault in CI before. Will have to dig into that, havent reproduced on any of my hardware yet.

azonenberg, 2025-07-10:
Aha, it's a bug in the AVX2 and AVX2+FMA implementations of the eye pattern filter that use an incorrect loop bound copied from the AVX512F version, which has a larger vector size. Fixed in https://github.com/ngscopeclient/scopehal/commit/9dd39728c0db1f926b67a6978b4fc75e7fed3bbd.

Both of my main dev machines are Xeons with AVX512F so it slipped through the cracks until I ran under asan (do we build CI unit tests builds with sanitizers? if not, we should, this would have caught it quickly... we should also make non-sanitizer release versions for end users to install though)

Gherkin, 2025-07-13:
Added another pull request for building in Arch Linux, like mentioned before. Everything builds and the tests run, both sdk and standalone.

#864

azonenberg, 2025-07-13:
PR merged, closing this issue

closed 2025-07-13 by azonenberg