ngscopeclient/scopehal-apps #868: High-DPI scaling is off

helaslo, 2025-07-18:
When setting the High-DPI scaling to 200% in the system settings, scaling inside ngscopeclient is off

Pictures of the issue:

[image]
[image]

KDE screen settings:

[image]

Output of ``ngscopeclient --debug --trace VulkanWindow``

```
OMP_WAIT_POLICY not set to PASSIVE
Re-exec'ing with correct environment
Initializing Vulkan
    VK_KHR_get_physical_device_properties2: supported
    VK_KHR_xcb_surface: supported
    VK_KHR_xlib_surface: supported
    VK_EXT_debug_utils: supported
    Loader/API support available for Vulkan 1.4
    Vulkan 1.2 support available, requesting it
    Initializing glfw 3.4.0 Wayland X11 GLX Null EGL OSMesa monotonic shared
    GLFW required extensions:
        VK_KHR_surface
        VK_KHR_wayland_surface
    Physical devices:
        Device 0: AMD Radeon Graphics (RADV GFX1152)
            API version:            0x00404137 (0.1.4.311)
            Driver version:         0x06401005 (0.25.1.5)
            Vendor ID:              1002
            Device ID:              1114
            Device type:            Integrated GPU
            int64:                  yes
            int16:                  yes (allowed in SSBOs)
            int8:                   yes (allowed in SSBOs)
            Max image dim 2D:       16384
            Max storage buf range:  4095 MB
            Max mem objects:        4294967295
            Max compute shared mem: 64 KB
            Max compute grp count:  4294967295 x 65535 x 65535
            Max compute invocs:     1024
            Max compute grp size:   1024 x 1024 x 1024
            Memory types:
                Type 0
                    Heap index: 1
                    Device local
                Type 1
                    Heap index: 1
                    Device local
                Type 2
                    Heap index: 0
                    Host visible
                    Host coherent
                Type 3
                    Heap index: 1
                    Device local
                    Host visible
                    Host coherent
                Type 4
                    Heap index: 1
                    Device local
                    Host visible
                    Host coherent
                Type 5
                    Heap index: 0
                    Host visible
                    Host coherent
                    Host cached
                Type 6
                    Heap index: 0
                    Host visible
                    Host coherent
                    Host cached
                Type 7
                    Heap index: 1
                    Device local
                    Device coherent
                    Device uncached
                Type 8
                    Heap index: 0
                    Host visible
                    Host coherent
                    Device coherent
                    Device uncached
                Type 9
                    Heap index: 1
                    Device local
                    Host visible
                    Host coherent
                    Device coherent
                    Device uncached
                Type 10
                    Heap index: 0
                    Host visible
                    Host coherent
                    Host cached
                    Device coherent
                    Device uncached
            Memory heaps:
                Heap 0
                    Size: 15 GB
                Heap 1
                    Size: 31 GB
                    Device local
        Device 1: llvmpipe (LLVM 20.1.7, 256 bits)
            API version:            0x00404137 (0.1.4.311)
            Driver version:         0x06401005 (0.25.1.5)
            Vendor ID:              10005
            Device ID:              0000
            Device type:            CPU
            int64:                  yes
            int16:                  yes (allowed in SSBOs)
            int8:                   yes (allowed in SSBOs)
            Max image dim 2D:       16384
            Max storage buf range:  128 MB
            Max mem objects:        4294967295
            Max compute shared mem: 32 KB
            Max compute grp count:  65535 x 65535 x 65535
            Max compute invocs:     1024
            Max compute grp size:   1024 x 1024 x 1024
            Memory types:
                Type 0
                    Heap index: 0
                    Device local
                    Host visible
                    Host coherent
                    Host cached
            Memory heaps:
                Heap 0
                    Size: 93 GB
                    Device local
        Selected device 0
            Queue families (5 total)
                Queue type 0
                    Queue count:          1
                    Timestamp valid bits: 64
                    Graphics
                    Compute
                    Transfer
                    Sparse binding
                Queue type 1
                    Queue count:          4
                    Timestamp valid bits: 64
                    Compute
                    Transfer
                    Sparse binding
                Queue type 2
                    Queue count:          1
                    Timestamp valid bits: 0
                Queue type 3
                    Queue count:          1
                    Timestamp valid bits: 0
                Queue type 4
                    Queue count:          1
                    Timestamp valid bits: 0
                    Sparse binding
            Driver: vk::DriverId::eMesaRadv
            Enabling 64-bit float support
            Enabling 64-bit integer support
            Enabling 16-bit integer support
            Enabling 16-bit integer support for SSBOs
            Enabling 8-bit integer support
            Enabling 8-bit integer support for SSBOs
            Device has VK_KHR_push_descriptor, requesting it
            Device has VK_KHR_shader_non_semantic_info, requesting it
            Device has VK_EXT_memory_budget, requesting it
            Device has VK_EXT_shader_atomic_float, requesting it
            Using heap 0, type 5 for pinned host memory
            Using heap 1, type 0 for card-local memory
            Sorted queues:
                Family=2 Index=0 Flags=00000020
                Family=3 Index=0 Flags=00000040
                Family=4 Index=0 Flags=00000008
                Family=1 Index=0 Flags=0000000e
                Family=1 Index=1 Flags=0000000e
                Family=1 Index=2 Flags=0000000e
                Family=1 Index=3 Flags=0000000e
                Family=0 Index=0 Flags=0000000f
            QueueManager creating family=1 index=0 name=g_vkTransferQueue
    
    vkFFT version: 1.3.3
Detecting CPU features...
    * AVX2
    * FMA
    * AVX512F
    * AVX512DQ
    * AVX512VL
    
Initializing locales
    Decimal separator is ,
Searching for plugins in /usr/lib/scopehal/plugins/
Searching for plugins in /usr/local/lib/scopehal/plugins/
Searching for plugins in /home/helaslo/randomTesztek/scopehal-apps/cmake-build-manual/src/ngscopeclient
Searching for plugins in /home/helaslo/.scopehal/plugins
QueueManager creating family=0 index=0 name=g_mainWindow.render
Using ImGui version 1.92.0
[VulkanWindow::UpdateFramebuffer] Recreating framebuffer due to window resize
[VulkanWindow::UpdateFramebuffer]     Max image extent: 16384 x 16384
[VulkanWindow::UpdateFramebuffer]     Current dims: 2560 x 1440
[VulkanWindow::VulkanWindow] Using ImGui style scale factor: 2.00
Vulkan driver is Mesa.
Disabling vkSetDebugUtilsObjectNameEXT on VkSurfaceKHR objects to work around driver bug.
QueueManager creating family=1 index=1 name=FilterGraphExecutor[0].queue
QueueManager creating family=1 index=2 name=FilterGraphExecutor[1].queue
QueueManager creating family=1 index=3 name=FilterGraphExecutor[3].queue

```

My system details:
```
Operating System: openSUSE Tumbleweed 20250717
KDE Plasma Version: 6.4.2
KDE Frameworks Version: 6.16.0
Qt Version: 6.9.1
Kernel Version: 6.15.6-1-default (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen AI 7 350 w/ Radeon 860M
Memory: 96 GiB of RAM (93.6 GiB usable)
Graphics Processor: AMD Radeon Graphics
Manufacturer: Framework
Product Name: Laptop 13 (AMD Ryzen AI 300 Series)
System Version: A7
```

azonenberg, 2025-07-18:
Keep an eye on https://github.com/ngscopeclient/scopehal-apps/issues/851 and let me know if things work better when it's fixed.

I'm pretty sure at least the filter graph title stuff will be improved there. But it does look like we might have a double scaling going on somewhere.

azonenberg, 2025-08-02:
Do you see any improvement or changes with a0a059c8216ab747a8d62ce3ebe6a07489d89cbe?

helaslo, 2025-08-03:
Sadly not much

[image]

Elemecca, 2025-09-17:
I can also reproduce on the current `master` version f253efd9c03911153b6ee5bb2bab052d8e674fda with gdb attached on my main dev machine, so I should be able to investigate in a few days when I have more free time.
- Arch Linux
- KDE Plasma on Wayland
- scaling set to 165% on a 27" 4K monitor
- AMD Radeon RX 9070 XT with the radv Vulkan driver

[image]

azonenberg, 2025-09-29:
Reopening: fixed on Linux and MacOS but still not working properly on Windows

helaslo, 2025-10-03:
For Linux only seems to be fixed on Wayland, if running via ``XDG_SESSION_TYPE=x11 ngscopeclient`` the scaling is still broken

Elemecca, 2025-10-04:
Interesting. It looks like [ImGui is ignoring the system DPI and setting its scaling factor to 1 on both macOS and Wayland](https://github.com/ngscopeclient/imgui/blob/a1c0836becbae04a0727d981b464f3a56867e2c1/backends/imgui_impl_glfw.cpp#L1061). It's probably not a coincidence that those are the platforms that currently work.

I've confirmed that, at least under a KDE Plasma Wayland session, XWayland _is_ setting the `Xft.dpi` X resource to the correct DPI value, and GLFW and ImGui are successfully reading it. I'm pretty sure that it's also correctly picking up the system scaling factor and activating ImGui's scaling on Windows.

It seems like we have these scaling problems whenever ImGui's scaling is active.

I think before #902 on Wayland we were actually double-scaling (using both ImGui's internal font scaling and whatever Wayland is doing natively) and that change just turned off the ImGui scaling, which resulted in correct rendering.