ngscopeclient/scopehal-apps #896: Binary v0.1 for Ubuntu 22.04 (Jammy Jellyfish) can not be installed due to libc6 version dependency.

CybEng, 2025-09-13:
The binary ngscopeclient-ubuntu-22.04-amd64-v0.1.deb for Ubuntu 22.04 (Jammy Jellyfish) can not be installed as it depends on libc6 version >=2.36-9   however the current official libc6 version for Ubuntu 22.04 is 2.35-0 per https://launchpad.net/ubuntu/jammy/+package/libc6.

azonenberg, 2025-09-13:
It compiles and runs fine on 22.04 but it looks like we hard coded 2.36-9 in the .deb generator code.

The full dependency list we have now is

```
set(CPACK_DEBIAN_PACKAGE_DEPENDS "libc6 (>= 2.36-9), libpng16-16(>= 1.6.39-2), libsigc++-2.0-0v5 (>= 2.12.0-1), libyaml-cpp0.7(>= 0.7.0), libgomp1(>= 12.2.0), libvulkan1(>= 1.3.239), libglfw3(>= 3.3.8), libgtk-3-0(>= 3.24.38), zlib1g(>= 1.2.13), libhidapi-hidraw0(>= 0.13.1), liblxi1(>= 1.18), libtirpc3 (>= 1.3.3)" )
```

Can you check what the correct versions of all of these for 22.04 is and I'll update? This list was generated on Debian Bookworm and I thought 22.04 was the same or older, but I guess not.. Oops.

We're planning a 0.1.1 maintenance release in the next couple weeks to fix some other small bugs found post release and I'll try to lump this in with it.

azonenberg, 2025-09-21:
@CybEng latest CI package as of 7ce3ba3 should install on 22.04. Can somebody look up the 24.04 package versions and comment here? I'll add that then we can close the ticket